TactilityProject/Tactility #151: Install/Getting Started

mg7410, 2025-01-09:
I’m familiar with this kind of project I understand it’s not ready to eat there’s adult assembly required. That’s part of the appeal for me there’s always something to learn along the way. But I’m sort of lost on where to start the install page says:

For now you can find some builds in GitHub, but they are only available for a few days or weeks after they were built:
* [Firmware builds](https://github.com/ByteWelder/Tactility/actions/workflows/build-firmware.yml)
* [SDK builds](https://github.com/ByteWelder/Tactility/actions/workflows/build-sdk.yml)
(scroll down to the bottom to find the zip files)
…
Run the flash script

Can’t find a zip link on those firmware/sdk pages, but regardless I get the idea whether its building or downloading I need to end up with firmware that will be compatible with the device I’m installing on and such.

Then to install this on a compatible ESP32 device the contents of the main directory and properly built firmware is sufficient? Or does it need the SDK as well? Once everything is ready and in the right place esptool and the flash script installs all the necessary components on the device.

That’s sort of what I’m thinking I’ve been looking around trying to better understand what role everything plays and how things interact. However just a basic idea of what needs to happen or whether I’m headed in the right direction or way off would be much appreciated.

Thanks!

KenVanHoeylandt, 2025-01-09:
Thanks for your interest in the project!

The builds are represented by the green checkmark as on the bottom-right in the screenshot:

[image]

Clicking on that gives you a popup:

[image]

This leads to a page that shows this:

[image]

Click on the link that says "Summary" at the top, with the house icon.

Then you land on a page where you scroll to the bottom and find the builds:

[image]

Make sure to **not** pick the download that has "-symbols" at the end.

KenVanHoeylandt, 2025-01-09:
If you use the flashing apps inside the zip, it will fully set up a device to run Tactility. Nothing additional is required, as it flashes all the required parts (program and data).