TactilityProject/Tactility #153: flash.ps1 flasher_args.json path

TechJeeper, 2025-01-10:
In the flash.ps1 the script looks for flasher_args.json in build when the extracted zip has it the directory named Binaries

KenVanHoeylandt, 2025-01-10:
Thanks for the report, I'll fix it.
I updated the scripts recently, but I didn't have a Windows machine to test it on. (I have a VM since today)

TechJeeper, 2025-01-10:
Here is my fixed flash.ps1 that I just used to flash my t-deck.  In Windows, it uses esptool.exe

----flash.ps1----

param(
    $port
)

# (Optional) Check if esptool.exe exists
if ((Get-Command "esptool.exe" -ErrorAction SilentlyContinue) -eq $null)
{
    Write-Host "Unable to find esptool.exe in your path. Make sure you have Python installed and on your path. Then run `pip install esptool`."
}

# Create flash command based on partitions
$json = Get-Content .\Binaries\flasher_args.json -Raw | ConvertFrom-Json
$jsonClean = $json.flash_files -replace '[\{\}\@\;]', ''
$jsonClean = $jsonClean -replace '[\=]', ' '

cd Binaries

# Corrected command with --connect-attempts
$command = "esptool.exe --connect-attempts 10 --port $port -b 460800 write_flash $jsonClean" 

# Or with full path to esptool.exe (recommended)
# $command = "C:\Path\To\Your\Python\Scripts\esptool.exe --connect-attempts 10 --port $port -b 460800 write_flash $jsonClean" 

Invoke-Expression $command
cd ..

TechJeeper, 2025-01-10:
Here is my fixed flash.ps1 that I just used to flash my t-deck. In Windows, it uses esptool.exe

`param(
    $port
)

# (Optional) Check if esptool.exe exists
if ((Get-Command "esptool.exe" -ErrorAction SilentlyContinue) -eq $null)
{
    Write-Host "Unable to find esptool.exe in your path. Make sure you have Python installed and on your path. Then run `pip install esptool`."
}

# Create flash command based on partitions
$json = Get-Content .\Binaries\flasher_args.json -Raw | ConvertFrom-Json
$jsonClean = $json.flash_files -replace '[\{\}\@\;]', ''
$jsonClean = $jsonClean -replace '[\=]', ' '

cd Binaries

# Corrected command with --connect-attempts
$command = "esptool.exe --connect-attempts 10 --port $port -b 460800 write_flash $jsonClean" 

# Or with full path to esptool.exe (recommended)
# $command = "C:\Path\To\Your\Python\Scripts\esptool.exe --connect-attempts 10 --port $port -b 460800 write_flash $jsonClean" 

Invoke-Expression $command
cd ..`

KenVanHoeylandt, 2025-01-10:
Thanks! I did something similar, but left the `.exe` out.
There were some other issues for me such as `--connect-attempts` not working, so I removed that.

New builds will be available in about 2 minutes here:
https://github.com/ByteWelder/Tactility/actions/runs/12718837697

Could you give one of those a try?

KenVanHoeylandt, 2025-01-11:
Fix was confirmed working. Thanks again for the report!

closed 2025-01-11 by KenVanHoeylandt