TactilityProject/Tactility #155: Can not build due to xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/semaphore_base

dienvx1997bn, 2025-01-11:
Hi,
I am trying to test your project with my ESP32S3 N8R2 board. Unfortunately, an error occurred the first time I ran `idf.py build`.
I suspect this is due to a bug in the kenel, or some macro was not being defined.
Anyone who has encountered this problem can help?
[image]

KenVanHoeylandt, 2025-01-11:
Are you using ESP-IDF v5.3.2?

(v5.4 was just published, but is possibly not compatible.)

DavidKarlas, 2025-01-11:
I also see this with 5.3.2, but I'm also on Windows.

KenVanHoeylandt, 2025-01-11:
Ah, I see. I don't have enough experience with Windows (or time to investigate) to fix the issue, but I'm open to contributions.

I did confirm it works with Win11+WSL+Ubuntu, but I do realize that's an entirely different approach.

DavidKarlas, 2025-01-11:
Getting same error with WSL, searching is not very useful, only good discussion is at https://github.com/wlav/cppyy-backend/issues/12 which mentions clang, being used which is not case for me :(

KenVanHoeylandt, 2025-01-11:
Interesting - good to know, thanks! It might be related to how filesystems inter-operate. (e.g. filesystem case sensitivity)

When I compiled from WSL, I also ran it from somewhere in the home directory.

closed 2025-02-03 by KenVanHoeylandt