TactilityProject/Tactility #156: Support for M5 Cardputer

vs4vijay, 2025-01-11:
Good to see this project exist, any plans to support M5 Cardputer ? Or any guidelines f someone wants to port to other ESP32 based board?

KenVanHoeylandt, 2025-01-11:
> Good to see this project exist, any plans to support M5 Cardputer ? 

I've been eye-ing the M5 Cardputer earlier, but I have no concrete plans yet for it.
The main challenge is the tiny screen and its 240x135 pixel resolution: the current interface patterns don't suit it well enough.

I intend to implement the [Lilygo T-Embed](https://lilygo.cc/products/t-embed-cc1101) this year, which will lead me to consider UI/UX on tiny screens, which __might__ pave the way for an M5 Cardputer implementation.

That said, Cardputer could be implemented already, it'd just be very tricky to use, because likely even the launcher icons won't fit. I'm thinking of a layout and/or resource swapping system like on Android, to resolve this.

> Or any guidelines f someone wants to port to other ESP32 based board?

Not yet, but I plan to expand documentation somewhere in the coming months.
If someone wants to port a port, it's best to look at the T-Deck implementation:
https://github.com/ByteWelder/Tactility/tree/main/Boards/LilygoTdeck

This is the main board configuration that references all the drivers and configuration bits for the T-Deck:
https://github.com/ByteWelder/Tactility/blob/main/Boards/LilygoTdeck/Source/LilygoTdeck.cpp

My `CONTRIBUTING.md` currently states that I'm not accepting contributions yet, but I'm definitely willing to make exceptions for boards and bug fixes. I'll update the document soon.

gwlvr, 2025-01-16:
man this would be so awesome..hope you can make it happen for cardputer fans!

NellowTCS, 2025-03-12:
Maybe a lighter version of the OS that is specifically for small screen devices?

KenVanHoeylandt, 2025-09-15:
The M5Stack Cardputer is now officially supported! More specifically: the original one and the V1.1 variant. 
The "Adv" variant is not yet supported, but it's very likely to become supported in the future.

 I still have some things to improve, but all the basics are working:
- Keyboard
- Display
- SD card
- USB Storage
- I2C/UART via "Port A"

I also made changes to the UI of Tactility to facilitate smaller screen devices. Some apps still need to be updated, but most of them work fine now.

Cardputer will be available from the web installer starting from the next release (0.6.0).
Meanwhile, you can browse the pipelines to find the latest firmware: 
1. Go to https://github.com/ByteWelder/Tactility/actions/workflows/build-firmware.yml
2. Make sure you're logged in on GitHub
3. Click on the top one in the list
4. Scroll down

[image]

closed 2025-09-15 by KenVanHoeylandt