TactilityProject/Tactility #160: No WiFi connect for M5stack-core2

rudyv, 2025-01-11:
Hi, successfully downloaded to my M5stack-core2, however I don't get a connection on my WiFi...
Any hints ?

I tried to download to a CYD ESP32 2432S028, no success

Great project !

KenVanHoeylandt, 2025-01-11:
Hi Rudy! Thanks for trying it out on your hardware!

I just flashed this firmware to my device and connected to WiFi on my Core2:
https://github.com/ByteWelder/Tactility/actions/runs/12724724479
(of course, entering credentials is super tricky - I am going to implemented the M5 I2C keyboard at some point)

Can you tell me how it fails exactly? Is it possible for you to run the serial monitor to see if there are errors? If not, can you open the Logs app and set it to filter warnings and up?

That said, I also have issues connecting to a specific WiFi SSIDs with _any_ ESP32 or S3 on another. It's the same router, just a different SSID, so it's something for me to investigate at some point.
I also have issues connecting to an iPhone that shares WiFi - I'm guessing the protocol is too modern there. It could be worth to check if your router is set to have classic b/g/n support or maybe only AX/AC support (over 2.4GHz)

`2432S028` is currently not supported.
I've only implemented `2432S024C`. The screens and touch drivers tend to be different across these boards.

rudyv, 2025-01-12:
Hi Ken, So I have a setup with Linksys Velops, I use the same said for 2.4 and 5 Ghz networks.  In the past, about +5 years ago the ESP8266 and ESP32 had issues with that, these days when I do something out of the Arduino IDE I don't have issue anymore. Would this thing maybe something that we experience ?  I'll test your latest build tomorrow

KenVanHoeylandt, 2025-01-12:
If your WiFi works with other ESP32, then I'm fairly certain that we can rule that out.
Is it perhaps possible to verify that WiFi on your Core2 does work with another piece of software?

Can you tell me which steps you take to connect to WiFi, and what happens and/or  doesn't happen?

KenVanHoeylandt, 2025-01-13:
Another thing to try is to erase flash and then flash. I've had weird bugs when NVS (Non-Volatile Storage, the place where I store WiFi settings) is broken because of incompatible formats.
The newer builds have a script that does this before flashing:

https://github.com/ByteWelder/Tactility/actions/runs/12735374245

rudyv, 2025-01-13:
Hi Ken, I just checked with a piece of Arduino code and the M5CORE2 comes online perfectly.

Then I have erased the flash and then programmed: ( based on https://github.com/ByteWelder/Tactility/actions/runs/12735374245 )

python3 -m esptool  --port /dev/cu.usbserial-020417B7 erase_flash      
python3 -m esptool  --port /dev/cu.usbserial-020417B7 write_flash 0x1000 bootloader.bin 0x10000 Tactility.bin 0x310000 system.bin 0x350000 data.bin 0x8000 partition-table.bin

Now the "WiFi Connect" is after said and password displaying the Tactility logo spinning all the time, it never stops.
At least we have different behavior.

Any hints ?

KenVanHoeylandt, 2025-01-13:
Thanks for investigating further! That does sound like the erase_flash step improved things a bit.

Could you open the apps folder and open the Log app? Then on the top right, you can filter for warnings. Is there anything wifi-related? (this gets erased upon reboot, so you have to try connect to WiFi first)

rudyv, 2025-01-13:
Hi Ken:
"Failed to set wifi config" ( ESP_ERR_WIFI_NOT_INIT)
wifi service: Connecting:Failed to disconnect ( ESP_ERR_WIFI_NOT_INIT)

KenVanHoeylandt, 2025-01-14:
That’s very useful! It likely has to do with the name or password of your WiFi.

Is there something special about your WiFi password or SSID name?
If you don’t want to discuss those details in public, you can find my email address via https://kenvanhoeylandt.net or find the Discord link on the https://tactility.one support page.
I might not have tested with an open WiFI (no password) so that could be it too.

KenVanHoeylandt, 2025-01-28:
I've pushed some WiFi changes that fixed something for me. Can you test it when you have time?
Binaries and installer are here: https://github.com/ByteWelder/Tactility/actions/runs/13015221346

KenVanHoeylandt, 2025-02-17:
Issue was verified to be fixed via email.

closed 2025-02-17 by KenVanHoeylandt