TactilityProject/Tactility #164: Cannot flash on tdeck plus on mac

mihirk98, 2025-01-12:
Hi,

Here is the output,

````
mihirk98@modeClient ~/tactility-lilygo-tdeck % ./flash.sh /dev/tty.usbmodem14201
-e \e[31m⚠️ jq not found!
	install jq from your package manager or install python and run 'pip install jq'\e[0m
./flash.sh: line 33: [[: /dev/tty.usbmodem14201: syntax error: operand expected (error token is "/dev/tty.usbmodem14201")
./flash.sh: line 40: jq: command not found
esptool.py v4.8.1
Serial port /dev/tty.usbmodem14201
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 48:ca:43:31:50:14
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.9s
Hard resetting via RTS pin...
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: the following arguments are required: <address> <filename>
mihirk98@modeClient ~/tactility-lilygo-tdeck % pip install jq
Requirement already satisfied: jq in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (1.8.0)
````

KenVanHoeylandt, 2025-01-13:
Thank you for the report, I'll look into this.

Meanwhile, you could try flashing via this https://esp.huhn.me/
It requires a browser that can has WebUSB/WebSerial, like Chrome, though.
It also requires the T-Deck to be in bootloader mode by keeping the trackball pressed and then press reset.

mihirk98, 2025-01-13:
Hi Ken,

Thank you for your reply, I am familiar with the website. Which file would I be flashing as there's multiple bin files?

KenVanHoeylandt, 2025-01-13:
All of these:
[image]

The offsets are in the zip file from the pipelines in `Binaries/flash_args`, but they are the same (currently) as in the screenshot.
I just tried it and it works (if you write `310000` and not `31000` :see_no_evil:)

KenVanHoeylandt, 2025-01-13:
Glad to hear it worked - thanks for reporting the issue!

Meanwhile, I've fixed the `flash.sh` script, so it should work now. I tested it on my old mac with Big Sur.
Would you be able to give it a try too?

The latest build is at the bottom of this page: https://github.com/ByteWelder/Tactility/actions/runs/12754487085

mihirk98, 2025-01-13:
works 👍🏼

KenVanHoeylandt, 2025-01-14:
Thank you!
FYI: There’s a web flasher now too: https://install.tactility.one

closed 2025-01-14 by KenVanHoeylandt