TactilityProject/Tactility #168: [M5stack Core S3] External i2c not working

PabloPL, 2025-01-15:
I'm trying to use [m5stack k](https://docs.m5stack.com/en/unit/cardkb_1.1) connected to external i2c port (port a), but it looks like it's not detecing anything connected to it. Tested with i've tried with i2c scanner app.

KenVanHoeylandt, 2025-01-15:
Thanks for the report! I likely broke it when I recently re-implemented the board.

Your suspicion about enabling power is probably right, because in the board re-implementation, a new driver was implemented for the power management IC.

KenVanHoeylandt, 2025-01-17:
I spent over 10 hours debugging this issue and failed to fix it so far.
I read the datasheets on the related power management ICs (AXP2101 and AW9523) and cross-referenced them with the schematics of the device, but I could not get it to work.

There must be a side-effect, or a time issue, or something else that I'm missing. 
I know M5Unified supposedly works, because the default demo firmware is based on that. The problem is that the initialization code of M5Unified is quite a [mess](https://github.com/m5stack/M5Unified/blob/b8cfec7fed046242da7f7b8024a4e92004a51ff7/src/M5Unified.cpp) so it's hard to figure out the exact bits and bytes of the correct init sequence.

I'm currently all out of ideas, but I'll revisit this later. Perhaps someone smarter than me can take a look at this in the meanwhile.

PabloPL, 2025-01-19:
'Funny' finding
- flash M5Stack CoreS3 UserDemo using m5burner
- observe that there is 5v on ext port and ext i2c is working fine
- flash Tactility (while device is still connected using usb cable)
- observe that there is 5v on ext port also (till next power on)

So maybe it's something/some register not being set.

KenVanHoeylandt, 2025-01-19:
Yep, I noticed that too. I even used that to figure out the initial state of the power management chip (after that initial flash in the "working" state) and make sure that I have the same state... to no avail.

PabloPL, 2025-01-26:
I think i got something - by looking at unified code, it looks like when doing i2c they don't wait for ACK from slave. And by just changing ACK_CHECK_EN from 1 to 0 i got i2c scanner correctly detect m5stack keyboard (https://shop.m5stack.com/products/cardkb-mini-keyboard-programmable-unit-v1-1-mega8a).
Maybe we could make it per board (or even per i2c device) ?

KenVanHoeylandt, 2025-01-26:
> I think i got something - by looking at unified code, it looks like when doing i2c they don't wait for ACK from slave. And by just changing ACK_CHECK_EN from 1 to 0 i got i2c scanner correctly [...]
Maybe we could make it per board (or even per i2c device) ?

Did you delete this comment? I got it via email, but I couldn't find it here.

I'm up for that if it fixes it! I could easily add it to the `tt::hal::i2c::Configuration`

Thanks for investigating further by the way, I really appreciate it!

PabloPL, 2025-01-26:
Yes i did deleted that comment because after restart it didn't worked. I've just forgot that it works if You flash first any m5stack code :( (for example platformio with m5unified in this case).

PabloPL, 2025-03-13:
It looks like someone already found a fix for this in https://github.com/espressif/esp-bsp/commit/214ce6f191fae75ae4b3872b295eebb58b9fd0a0.
I've created PR with similar change https://github.com/ByteWelder/Tactility/pull/246

KenVanHoeylandt, 2025-03-13:
Thank you for the contribution!

closed 2025-03-13 by KenVanHoeylandt