TactilityProject/Tactility #170: ESP32-2432S022C Battery Support

NellowTCS, 2025-01-19:
Do you have support for using a battery with the ESP32-2432S022C (CYD)?
I saw that "Power" isn't compatible but don't know what you mean. I have attached some images of the CYD below. </n>
[image]</n>
[image]</n>
[image]</n>
[image]</n>
[image]</n>

KenVanHoeylandt, 2025-01-19:
Note that the supported board is 2432S0**24C** and not 2432S0**22C**.

If the latter has the exact same hardware configuration as the former, then it will work. I don't own the 22C variant, so I can't test it.
The physical size of the screen won't be problematic, but if the connected circuits and pins are different, then it might not work.

The battery connector should work fine on any of the boards.

The support for "power" actually means support for "power status". In other words, things like the battery indicator, and the info in the Power settings app.
I've updated the website to say "power status" instead.

I've looked into the [schematics](https://github.com/memchi/ESP32-2432S024/tree/main/Original) and couldn't find any power measurement chips or setup (e.g. voltage divider), hence the feature is marked as unavailable. I'm not super good at reading schematics, so if someone else has a tip on measuring power, then I'm all ears.

NellowTCS, 2025-01-19:
Oh whoops! 
Thanks for the help.
While digging around I found that the provided link for this board's materials is: http://pan.jczn1688.com/s/j1fgva
I downloaded it and, while looking in the schematic found this:

[image]</n>

[image]

KenVanHoeylandt, 2025-01-19:
I saw that. The battery charging circuitry only does that. If I read it correctly, there are no pins for voltage or current sensing, nor are there ICs that do power management to communicate that via I2C.
There might be an indirect way to read voltage (and thus battery power remaining can be roughly estimated) from some other component (e.g. display), but I haven't gone very deep into the schematics.

NellowTCS, 2025-01-19:
Okay thanks for the help!
I'll try to dig through the files and find something!

KenVanHoeylandt, 2025-01-22:
I'll close this issue, but it can be re-opened if someone figures out a way to read the battery voltage.

closed 2025-01-22 by KenVanHoeylandt