TactilityProject/Tactility #171: Idea for Web Browser

NellowTCS, 2025-01-19:
This is a bit of a weird idea but is it possible to implement something like
https://www.instructables.com/Google-Search-on-ESP32/
as an app for this?

KenVanHoeylandt, 2025-01-19:
I recently started considering to build a simple web browser, but concluded it was too much work and the result would be disappointing regardless.

While the example does work, it's super limited:
- You have to register an API key: Every user of the firmware would have to create their own API key as these keys aren't intended to be shared. It's quite a hassle.
- You can't really do anything with the search results because there isn't a web browser to open them.

I think dedicated apps for certain services might work better, like a WikiPedia or translation app.

NellowTCS, 2025-01-19:
Yeah that would be a better idea.

NellowTCS, 2025-04-02:
I am working on something:
https://github.com/NellowTCS/Tactility/blob/main/App/Source/TactileWeb/TactileWeb.cpp

closed 2025-07-23 by NellowTCS