TactilityProject/Tactility #175: Possible new device CL-32

moosepr, 2025-01-20:
Ahoy there, I have been working on a new device, and came across your firmware and wondered how well the 2 would play.

I have been padding out the website [here](http://cl-32.com) but the TL:DR its an ESP32-S3 with a 384x168 e-ink display, 72 key keyboard, and a custom add on slot.

Im not sure how well your UI would fit with the screen, and i dont have a touch interface.

How do you think it will work?

Pete

KenVanHoeylandt, 2025-01-20:
Cool project!

The main challenge would be the UI/UX, as I don't expect it to work well on 384x168 pixels with the current state of things. 
The absence of a touch screen doesn't have to be a problem if we can figure an alternative way for navigating (e.g. WSAD for interface element selection)

That said, I was already thinking about how to support devices with smaller screens and the absence of touch input.

I intend to implement the [Lilygo T-Embed CC1101](https://lilygo.cc/products/t-embed-cc1101) this year: this device has a 320x170 pixel display. If this is a success, then the UX will be sufficiently adapted to work on your screen too.
I'm thinking of:
- Making the status bar (black top bar) thinner
- Making the toolbar thinner
- Implement alternative controls
- Consider loading alternative layouts in apps for smaller screen device (a bit like on Android)

Additionally, I'll also be implementing the [T-Deck Pro](https://github.com/Xinyuan-LilyGO/T-Deck-Pro) soon-ish, so that will be the first venturing for me into e-ink displays. That should give me some experience with slower refresh rates.

NellowTCS, 2025-11-05:
Major success!!!!!!
(yes a fix for the shift is compiling right now)

https://github.com/user-attachments/assets/5dbde633-3a3e-4647-902c-6edff0195d15

NellowTCS, 2025-11-06:
[image]
[image]
[image]

It renders super cleanly (I will make those fonts work better later), but the display is rendered twice, and no matter what I do, I cannot get it to stop doing that.

If either of you has any idea why this happens, let me know, as I'm rather stumped!

KenVanHoeylandt, 2025-11-06:
I'd first verify that the flush call is moving the correct amount of bytes. Then I'd verify that any pixel format conversion is done right.
Assuming the Y axis is correctly rendered, it could be that the X axis is skipping to render a pixel every few pixels. 

I can probably help debug this over the weekend.

NellowTCS, 2025-11-06:
> I'd first verify that the flush call is moving the correct amount of bytes. Then I'd verify that any pixel format conversion is done right. Assuming the Y axis is correctly rendered, it could be that the X axis is skipping to render a pixel every few pixels.

I've checked these, and they seem to be okay, but the main issue I suspect is related to my stride logic. However, I'm not sure what else could be causing the problem, as I've tried a lot of things.

> I can probably help debug this over the weekend.

That would be amazing, thank you!!

moosepr, 2025-11-06:
I probably cant contribute much as I dont understand all the screen code, but I did remember a library I was using early on. It didnt quite support the screen I was using at the time, and it uses its own SPI back end code, so I struggled to get the SD card working with the shared clock/data pins with the screen, but maybe it has some info that helps? 

https://github.com/bitbank2/bb_epaper

There is also some sample code from the manufacturer, no idea if that is any help either?

[A32-GDEY029T71H.zip](https://github.com/user-attachments/files/23389263/A32-GDEY029T71H.zip)

NellowTCS, 2025-11-08:
I found a driver for this display, and best part is that it's esp-idf compatible!
https://github.com/NellowTCS/Tactility/tree/refs/heads/cl-32-only-ssd1680

I currently am getting it working in another branch:
https://github.com/NellowTCS/Tactility/tree/cl-32-only-ssd1680

NellowTCS, 2026-10-10:
I believe this is completed @KenVanHoeylandt!

KenVanHoeylandt, 2026-10-10:
Yep, thanks for bringing this to my attention!

closed 2026-10-10 by KenVanHoeylandt