TactilityProject/Tactility #189: Memory pressure issue when enabling WiFi on device without PSRAM

KenVanHoeylandt, 2025-01-25:
ESP32 (CYD) memory issues (or any device without PSRAM). When you enable WiFi and it connects:
  - Boot app doesn't show logo 
  - Desktop makes desktop icons disappear
  - WiFi might fail quietly when trying to enable it: this shows no feedback (force it by increasing LVGL buffers to 100kB)

Possible mitigations: 
  - When no PSRAM is available, use simplified desktop buttons
  - Add statusbar icon for memory pressure.
  - Show error in WiFi screen (e.g. AlertDialog when SPI is not enabled and available memory is below a certain amount)

NellowTCS, 2025-03-22:
These issues usually don't appear if you don't plug in an MicroSD card.

KenVanHoeylandt, 2026-09-01:
Seems to be resolved.

Only the Elecrow 3.5" devices have an issue still, but I'm closing this for now. The Elecrow issues are mentioned in their device.properties file.

closed 2026-09-01 by KenVanHoeylandt