TactilityProject/Tactility #190: M5Stack only showing 4MB of SPIRAM/PSRAM

KenVanHoeylandt, 2025-01-25:
Core2 or CoreS3 (TODO: check this) only shows 4MB of SPIRAM while it should show 8MB

PabloPL, 2025-01-28:
On CoreS3, on System info it's showing SPI memory usage 194 / 8192 kB so it looks like it's fine.

KenVanHoeylandt, 2025-01-28:
Thanks, I'll update the title then!

NellowTCS, 2025-03-29:
I found a reason as to why this might be happening:
"Older ESP32s can only map up to 4MB RAM.
https://www.esp32.com/viewtopic.php?t=12710
Turn on your logging and you’ll probably see this output as the ESP 32 boots up.
`esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped` "

"You can access the rest via the himem API which allows you to page in chunks of the other 4M a little bit at a time. Think of it like very fast swap space"

https://www.reddit.com/r/esp32/comments/11du8iz/esp32_wrover_kit_why_only_4mb_psram_data_sheet/

KenVanHoeylandt, 2025-03-29:
Thanks for investigating and figuring this out! I'll add it to the documentation for that device soon.

KenVanHoeylandt, 2025-04-01:
It's updated: https://tactility.one/#/devices/m5stack-core2

closed 2025-04-01 by KenVanHoeylandt