TactilityProject/Tactility #196: Adding a simple PNG to the Files -> data folder

mihirk98, 2025-01-29:
Hi Byte,

We've spoken before, I asked for help regarding flashing Tactility onto a T-Deck Plus. Going really slowly to understanding and making changes to the system to suit my needs, I wanted to understand how to do something simple like adding a png file to Files -> data folder.

I can see where the data folder is in the code. I can add my file there (and assuming like other pngs in the system folder, when I click on it, it will simply open up and show up on the T-Deck like other png files (albeit these other png files being in a different sub-folder i.e. mainly the system sub-folder)

Now, while I can add my file simply, how would I go about building the bin files to flash my t-deck with this new version of the system (the only change being adding a png file, I do not have an sd-card as on now so that's not an option yet). Am I going about the right way with this, is there an easier option?

Thank you for your time, look forward to hearing from you :)

mihirk98, 2025-01-29:
I'm assuming I simply need to add the file to the Data -> data folder but then how do I convert it to a .bin file? I know this- "0x350000 data.bin" is where I flash it. So after I manage to convert, I can simply and only flash at that location and I should be good to go?

KenVanHoeylandt, 2025-02-01:
Due to the filesystem being case-insensitive on macOS, the POSIX include `semaphore.h` was preferred over `Semaphore.h` from Tactility.
On Linux with a case-sensitive filesystem, those two files are considered different, hence it was only failing on a standard macOS setup.

I moved all Tactility headers into a `Tactility/` folder, so `Semaphore.h` is now included as `Tactility/Semaphore.h`.

closed 2025-02-01 by KenVanHoeylandt