TactilityProject/Tactility #197: ESP32-2432S022C Support

NellowTCS, 2025-01-30:
So I tried to flash the ESP32-2432S024C  (2.4" CYD) image onto the ESP32-2432S022C (2.2" CYD) and got a white screen with pressing Boot turning the screen off and on.
I did some research and fiddled with the code a bit but nothing worked.
- Apparently the 2.2" CYD has a ST7789 display driver, so you cannot use the code for the ILI9341 based CYDs.
- it uses the Parallel interface (i80) between display and ESP32
- has GPIO 27 for the TFT, instead of GPIO 21

I found a LVGL example which was made for the 2.2" CYD however so this may be helpful!  
https://github.com/Risto97/ESP32-2432S022_lvgl_example

I am willing to try to implement the board myself but would need a tiny bit of help (just what changes needed and locations for changes) as I am not the best at C/C++.

Thanks!

KenVanHoeylandt, 2025-02-01:
I recently wrote a basic guide on how to develop a new board:
https://docs.tactility.one/#/board-development

It's not detailed, but it gives you an idea of the things to consider. (all docs are work in progress)

The main limiting factor for your board will be RAM. Connecting with WiFi will likely make the desktop icons and other icons disappear as there isn't enough memory left. If you don't turn on WiFi, everything else should be just fine.
This can be fixed by optimizing both WiFi (buffer sizes) and desktop icon sizes (make them smaller), but I haven't given that any priority yet.

If you like, you can join Discord and open a thread there to discuss, but we can also do that here.

There's also this: https://github.com/ByteWelder/Tactility/blob/main/CONTRIBUTING.md
tl;dr I currently won't merge code into the main repository for boards that I don't own, but I'm willing to link to your fork from the main website.

NellowTCS, 2025-02-01:
Alright thanks!

NellowTCS, 2025-02-09:
@Shadowtrance 
This is amazing thanks for the help!

msahin, 2025-02-26:
is there any build for **2432s022c**

KenVanHoeylandt, 2025-02-26:
@msahin Not currently, as I don't own that device. The supported hardware is listed here: https://tactility.one/#/supported-devices

NellowTCS, 2025-03-13:
> is there any build for **2432s022c**

> [@msahin](https://github.com/msahin) Not currently, as I don't own that device. The supported hardware is listed here: https://tactility.one/#/supported-devices

I am working on developing that board in a fork: https://github.com/NellowTCS/Tactility

KenVanHoeylandt, 2025-03-13:
Feel free to open a pull request when the implementation works.

NellowTCS, 2025-03-14:
Alright thanks!

NellowTCS, 2025-03-14:
So do you have any idea why I got a white screen (noticeably brighter than the firmware for CYD-2432S024C however) on boot on the latest build of the fork I made (attached below)?

[Tactility-cyd-2432S022c.zip](https://github.com/user-attachments/files/19239534/Tactility-cyd-2432S022c.zip)

KenVanHoeylandt, 2025-03-15:
Do you have screenshots or a short video?

NellowTCS, 2025-03-16:
Okay so I just saw this, but I did fix the issue but had the same result (white screen).
Previously, the serial terminal showed:
``` 
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:6592

load:0x40078000,len:16656

load:0x40080400,len:4

load:0x40080404,len:4268

entry 0x40080658

I (29) boot: ESP-IDF v5.4 2nd stage bootloader

I (29) boot: compile time Mar 14 2025 01:20:48

I (29) boot: Multicore bootloader

I (31) boot: chip revision: v3.1

I (33) qio_mode: Enabling default flash chip QIO

I (38) boot.esp32: SPI Speed      : 40MHz

I (41) boot.esp32: SPI Mode       : QIO

I (45) boot.esp32: SPI Flash Size : 4MB

I (49) boot: Enabling RNG early entropy source...

I (53) boot: Partition Table:

I (56) boot: ## Label            Usage          Type ST Offset   Length

I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (75) boot:  2 factory          factory app      00 00 00010000 00300000

I (82) boot:  3 system           Unknown data     01 81 00310000 00040000

I (88) boot:  4 data             Unknown data     01 81 00350000 00040000

I (95) boot: End of partition table

I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=5d978h (383352) map

I (216) esp_image: segment 1: paddr=0006d9a0 vaddr=3ff80000 size=0001ch (    28) load

I (216) esp_image: segment 2: paddr=0006d9c4 vaddr=3ffb0000 size=02654h (  9812) load

I (223) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=13ff00h (1310464) map

I (605) esp_image: segment 4: paddr=001aff28 vaddr=3ffb2654 size=03394h ( 13204) load

I (610) esp_image: segment 5: paddr=001b32c4 vaddr=40080000 size=15080h ( 86144) load

I (650) boot: Loaded app from partition at offset 0x10000
[... 363 lines cut ...]
I (916) lvgl_init: Starting

I (919) system_event: BootInitLvglBegin

I (923) LVGL: Starting LVGL task

E (926) lcd_panel.io.i80: esp_lcd_new_i80_bus(164): no mem for format buffer

ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x400dfeea

file: "./Boards/CYD-2432S022C/Source/hal/YellowDisplay.cpp" line 50

func: CustomI80Display::CustomI80Display(std::shared_ptr<tt::hal::touch::TouchDevice>)

expression: esp_lcd_new_i80_bus(&bus_config, &i80_bus)

abort() was called at PC 0x4008a86b on core 0

Backtrace: 0x40081c55:0x3ffbe720 0x4008a875:0x3ffbe740 0x4008f0e9:0x3ffbe760 0x4008a86b:0x3ffbe7d0 0x400dfeea:0x3ffbe800 0x400e0174:0x3ffbe950 0x400e01a4:0x3ffbe980 0x400e01ca:0x3ffbe9b0 0x400e01df:0x3ffbe9d0 0x400f18aa:0x3ffbea10 0x400f199e:0x3ffbea40 0x400e098d:0x3ffbea90 0x400de6fe:0x3ffbeac0 0x4020bd9a:0x3ffbeaf0 0x4008ac6a:0x3ffbeb20

ELF file SHA256: 91e401042
```
The main issue is that nothing ever displays on the screen, only a white blank screen and after fixing the memory issue, the white screen shows and then the screen turns off.

(looks the same as the previous image)

Shadowtrance, 2025-03-16:
My suggestion would be to go back to basics and start with an idf example which they have actually provided [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i80_controller) to get a working baseline and familiarity with getting it going with esp-idf and lvgl_port.

Then transfer what you figured out with that config to get it going in tactility with little to no issue. And it's way faster to test, fix, test..... that way too.

That's what i do anyway if i haven't used a display with idf yet and don't have a config i can pull from in another project. :)

KenVanHoeylandt, 2025-03-16:
Sidenote: I changed the way boards are registered. Instead of editing `CMakeLists.txt` and `App/CMakeLists.txt`, boards are now defined in `Buildscripts/board.cmake`.

NellowTCS, 2025-03-16:
> My suggestion would be to go back to basics and start with an idf example which they have actually provided [...]

So I checked that example out and it has the same issue as mine (white screen on boot). While searching about it I found that the TFT_SPI library nor ST8879 libraries work well with this display, only LovyanGFX works well. Thanks for the link!

NellowTCS, 2025-03-16:
> Sidenote: I changed the way boards are registered. Instead of editing `CMakeLists.txt` and `App/CMakeLists.txt`, boards are now defined in `Buildscripts/board.cmake`.
> 
> You'll likely get a merge conflict when you update your fork. Just revert the `CMakeLists.txt` changes and update the `board.cmake` file.

I am starting from scratch now but thanks for the note!

NellowTCS, 2025-03-16:
> Sidenote: I changed the way boards are registered. Instead of editing `CMakeLists.txt` and `App/CMakeLists.txt`, boards are now defined in `Buildscripts/board.cmake`.
>
> You'll likely get a merge [...]

So just making sure, 
Neither of the below steps are now necessary?
"[Configure App project](https://docs.tactility.one/#/board-development?id=configure-app-project)
In App/, add your board as a dependency in the CMakeLists.txt.

Update the Boards.h to include your board."

KenVanHoeylandt, 2025-03-17:
@NellowTCS Apparenlty I forgot to update that part of the docs, I'll do that in a few minutes:

This part isn't necessary anymore:

> In App/, add your board as a dependency in the CMakeLists.txt.

This part is still necessary:

> Update the Boards.h to include your board.

NellowTCS, 2025-03-18:
So currently the firmware is building and flashing but I get this (decoded) backtrace after the display init begins:
0x40114442: task_wdt_timeout_handling at /opt/esp/idf/components/esp_system/task_wdt/task_wdt.c:434
0x4011483c: task_wdt_isr at /opt/esp/idf/components/esp_system/task_wdt/task_wdt.c:507
0x400e1769: lv_display_set_buffers at /app/NellowTCS/Tactility/Libraries/lvgl/src/display/lv_display.c:444
0x400d98c4: LovyanGFXDisplay::start() at /app/NellowTCS/Tactility/Boards/CYD-2432S022C/Source/hal/LovyanDisplay.cpp:115
0x400d909f: app_main at /app/NellowTCS/Tactility/Boards/CYD-2432S022C/Source/CYD2432S022C.cpp:110
0x4012c4b9: main_task at /opt/esp/idf/components/freertos/app_startup.c:208
0x4008618e: vPortTaskWrapper at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

here are the logs:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:6592

load:0x40078000,len:16656

load:0x40080400,len:4

load:0x40080404,len:4268

entry 0x40080658

I (29) boot: ESP-IDF v5.4 2nd stage bootloader

I (29) boot: compile time Mar 18 2025 19:01:59

I (29) boot: Multicore bootloader

I (31) boot: chip revision: v3.1

I (33) qio_mode: Enabling default flash chip QIO

I (38) boot.esp32: SPI Speed      : 40MHz

I (41) boot.esp32: SPI Mode       : QIO

I (45) boot.esp32: SPI Flash Size : 4MB

I (49) boot: Enabling RNG early entropy source...

I (53) boot: Partition Table:

I (56) boot: ## Label            Usage          Type ST Offset   Length

I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (75) boot:  2 factory          factory app      00 00 00010000 00300000

I (82) boot:  3 system           Unknown data     01 81 00310000 00040000

I (88) boot:  4 data             Unknown data     01 81 00350000 00040000

I (95) boot: End of partition table

I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=13708h ( 79624) map

I (128) esp_image: segment 1: paddr=00023730 vaddr=3ff80000 size=0001ch (    28) load

I (129) esp_image: segment 2: paddr=00023754 vaddr=3ffb0000 size=02fc4h ( 12228) load

I (136) esp_image: segment 3: paddr=00026720 vaddr=40080000 size=098f8h ( 39160) load

I (153) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=5e4d0h (386256) map

I (265) esp_image: segment 5: paddr=0008e4f8 vaddr=400898f8 size=00c3ch (  3132) load

I (271) boot: Loaded app from partition at offset 0x10000

I (271) boot: Disabling RNG early entropy source...

I (283) cpu_start: Multicore app

I (291) cpu_start: Pro cpu start user code

I (291) cpu_start: cpu freq: 240000000 Hz

I (291) app_init: Application information:

I (291) app_init: Project name:     Tactility

I (295) app_init: App version:      0.3.0

I (299) app_init: Compile time:     Mar 18 2025 19:02:30

I (304) app_init: ELF file SHA256:  38920a1a9...

I (308) app_init: ESP-IDF:          v5.4

I (312) efuse_init: Min chip rev:     v0.0

I (316) efuse_init: Max chip rev:     v3.99 

I (320) efuse_init: Chip rev:         v3.1

I (324) heap_init: Initializing. RAM available for dynamic allocation:

I (330) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM

I (335) heap_init: At 3FFBF2C8 len 00020D38 (131 KiB): DRAM

I (340) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM

I (345) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM

I (351) heap_init: At 4008A534 len 00015ACC (86 KiB): IRAM

I (357) spi_flash: detected chip: generic

I (360) spi_flash: flash io: qio

W (363) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`

I (374) main_task: Started on CPU0

I (377) main_task: Calling app_main()

I (380) LovyanDisplay: Starting LovyanGFX display

E (5581) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:

E (5581) task_wdt:  - IDLE0 (CPU 0)

E (5581) task_wdt: Tasks currently running:

E (5581) task_wdt: CPU 0: main

E (5581) task_wdt: CPU 1: IDLE1

E (5581) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x40114442:0x3FFB1C00 0x4011483C:0x3FFB1C20 0x40082849:0x3FFB1C50 0x400E1769:0x3FFC0530 0x400D98C4:0x3FFC0570 0x400D909F:0x3FFC0590 0x4012C4B9:0x3FFC05C0 0x4008618E:0x3FFC05F0

E (10581) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:

E (10581) task_wdt:  - IDLE0 (CPU 0)

E (10581) task_wdt: Tasks currently running:

E (10581) task_wdt: CPU 0: main

E (10581) task_wdt: CPU 1: IDLE1

E (10581) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x40114442:0x3FFB1C00 0x4011483C:0x3FFB1C20 0x40082849:0x3FFB1C50 0x400E1769:0x3FFC0530 0x400D98C4:0x3FFC0570 0x400D909F:0x3FFC0590 0x4012C4B9:0x3FFC05C0 0x4008618E:0x3FFC05F0

E (15581) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:

E (15581) task_wdt:  - IDLE0 (CPU 0)

E (15581) task_wdt: Tasks currently running:

E (15581) task_wdt: CPU 0: main

E (15581) task_wdt: CPU 1: IDLE1

E (15581) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x40114442:0x3FFB1C00 0x4011483C:0x3FFB1C20 0x40082849:0x3FFB1C50 0x400E1769:0x3FFC0530 0x400D98C4:0x3FFC0570 0x400D909F:0x3FFC0590 0x4012C4B9:0x3FFC05C0 0x4008618E:0x3FFC05F0
```
I believe this is due to either not having enough memory or some processes not being ended or started correctly, but just wanted to see if any of you have faced the same issue before and know exactly what the problem is.
Thanks for the help though!

KenVanHoeylandt, 2025-03-18:
The watchdog timer checks that tasks aren't hanging endlessly on an operation. Idle tasks are never problematic, but in this case the main task on CPU 0 is stuck/stalling.

The problem is likely somewhere here:

> 0x400e1769: lv_display_set_buffers at /app/NellowTCS/Tactility/Libraries/lvgl/src/display/lv_display.c:444
> 0x400d98c4: LovyanGFXDisplay::start() at /app/NellowTCS/Tactility/Boards/CYD-2432S022C/Source/hal/LovyanDisplay.cpp:115

If we look at `lv_display_set_buffers()`:

[image]

It's likely that this assert is being triggered in a loop.

NellowTCS, 2025-03-18:
That's super helpful thanks!

NellowTCS, 2025-03-19:
Okay so I know the error:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:6592

load:0x40078000,len:16656

load:0x40080400,len:4

load:0x40080404,len:4268

entry 0x40080658

I (29) boot: ESP-IDF v5.4 2nd stage bootloader

I (29) boot: compile time Mar 19 2025 01:20:19

I (29) boot: Multicore bootloader

I (31) boot: chip revision: v3.1

I (33) qio_mode: Enabling default flash chip QIO

I (38) boot.esp32: SPI Speed      : 40MHz

I (41) boot.esp32: SPI Mode       : QIO

I (45) boot.esp32: SPI Flash Size : 4MB

I (49) boot: Enabling RNG early entropy source...

I (53) boot: Partition Table:

I (56) boot: ## Label            Usage          Type ST Offset   Length

I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (75) boot:  2 factory          factory app      00 00 00010000 00300000

I (82) boot:  3 system           Unknown data     01 81 00310000 00040000

I (88) boot:  4 data             Unknown data     01 81 00350000 00040000

I (95) boot: End of partition table

I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=13ce8h ( 81128) map

I (129) esp_image: segment 1: paddr=00023d10 vaddr=3ff80000 size=0001ch (    28) load

I (129) esp_image: segment 2: paddr=00023d34 vaddr=3ffb0000 size=02fc4h ( 12228) load

I (137) esp_image: segment 3: paddr=00026d00 vaddr=40080000 size=09318h ( 37656) load

I (153) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=5e7a0h (386976) map

I (265) esp_image: segment 5: paddr=0008e7c8 vaddr=40089318 size=0121ch (  4636) load

I (272) boot: Loaded app from partition at offset 0x10000

I (272) boot: Disabling RNG early entropy source...

I (283) cpu_start: Multicore app

I (291) cpu_start: Pro cpu start user code

I (291) cpu_start: cpu freq: 240000000 Hz

I (291) app_init: Application information:

I (291) app_init: Project name:     Tactility

I (295) app_init: App version:      0.3.0

I (299) app_init: Compile time:     Mar 19 2025 01:20:50

I (304) app_init: ELF file SHA256:  6edc1e0ba...

I (308) app_init: ESP-IDF:          v5.4

I (312) efuse_init: Min chip rev:     v0.0

I (316) efuse_init: Max chip rev:     v3.99 

I (320) efuse_init: Chip rev:         v3.1

I (324) heap_init: Initializing. RAM available for dynamic allocation:

I (330) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM

I (335) heap_init: At 3FFC2EC8 len 0001D138 (116 KiB): DRAM

I (340) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM

I (346) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM

I (351) heap_init: At 4008A534 len 00015ACC (86 KiB): IRAM

I (358) spi_flash: detected chip: generic

I (360) spi_flash: flash io: qio

W (363) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`

I (374) main_task: Started on CPU0

I (377) main_task: Calling app_main()

I (379) CYD2432S022C: Tactility v0.3.0 starting from CYD2432S022C.cpp

I (386) CYD2432S022C: LVGL initialized from CYD2432S022C.cpp

I (391) LovyanDisplay: Creating display

I (394) LovyanDisplay: Display configuration created

I (399) LovyanDisplay: Initializing LGFX_CYD_2432S022C display class

I (405) LovyanDisplay: Configuring LCD bus pins

I (409) LovyanDisplay: LCD bus configuration applied successfully

I (415) LovyanDisplay: Configuring LCD panel

I (419) LovyanDisplay: LCD panel configuration applied successfully

I (425) LovyanDisplay: Configuring backlight settings

I (430) LovyanDisplay: Backlight configured successfully

I (435) LovyanDisplay: LovyanGFXDisplay constructor called

I (440) LovyanDisplay: Starting LovyanGFX display

I (655) LovyanDisplay: LCD init called

I (705) LovyanDisplay: Display initialized with brightness=128 and rotation=0

I (710) LovyanDisplay: LVGL display create called

I (720) LovyanDisplay: LVGL display created: 320x240

I (720) LovyanDisplay: Buffer size = 10240 pixels, 30720 bytes (LV_COLOR_DEPTH=16, sizeof(lv_color_t)=3)

I (722) LovyanDisplay: Calling lv_display_set_buffers with size=10240

E (5718) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:

E (5718) task_wdt:  - IDLE0 (CPU 0)

E (5718) task_wdt: Tasks currently running:

E (5718) task_wdt: CPU 0: main

E (5718) task_wdt: CPU 1: IDLE1

E (5718) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x40114712:0x3FFB1C00 0x40114B0C:0x3FFB1C20 0x40082849:0x3FFB1C50 0x400E1A39:0x3FFC40E0 0x400D9A63:0x3FFC4120 0x400D913F:0x3FFC4150 0x4012C789:0x3FFC4180 0x4008618E:0x3FFC41B0
```

LVGL (and I believe ESP_LVGL_Port (even though it has been removed from my board's CMakeLists.txt) keep setting the LV_COLOR_DEPTH to 32 and lv_color_t to 3 , while apparently this board should use LV_COLOR_DEPTH as 16 and lv_color_t as 2. Also, due to this, the buffer keeps being set to  30720 and not 20,480. 
I did try using a manual config for lvgl, but so far this isn't working.

NellowTCS, 2025-03-20:
Update: I decided to manually set the buffers without using lv_color_t and this works.

NellowTCS, 2025-03-20:
Sorry for the bad image but I got output!

[image]

NellowTCS, 2025-03-21:
Okay, good news! I got the display output and OS working!!!

Only one problem though, that rotation doesn't work and leads to weird screen glitching on switching orientation in Settings. I took a peek at the other configurations and they don't have any special rotation settings. 
Are rotation settings handled by the system (and so do I not need to not set orientation in my display config) and, if not, how would I need to modify my current implementation to make it work?
Current Display Configuration:
```
#include "LovyanDisplay.h"
#include "CYD2432S022CConstants.h"
#include "CST820Touch.h"
#include <esp_log.h>
#include <LovyanGFX.h>
#include <inttypes.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_heap_caps.h>

static const char* TAG = "LovyanDisplay";

// Define a custom LovyanGFX class for the CYD-2432S022C board
class LGFX_CYD_2432S022C : public lgfx::LGFX_Device {
public:
    lgfx::Panel_ST7789 _panel_instance;
    lgfx::Bus_Parallel8 _bus_instance;
    lgfx::Light_PWM _light_instance;

    LGFX_CYD_2432S022C() {
        ESP_LOGI(TAG, "Initializing LGFX_CYD_2432S022C display class");
        
        // Bus configuration
        {
            auto cfg = _bus_instance.config();
            cfg.pin_wr = CYD_2432S022C_LCD_PIN_WR;
            cfg.pin_rd = CYD_2432S022C_LCD_PIN_RD;
            cfg.pin_rs = CYD_2432S022C_LCD_PIN_DC;
            cfg.pin_d0 = CYD_2432S022C_LCD_PIN_D0;
            cfg.pin_d1 = CYD_2432S022C_LCD_PIN_D1;
            cfg.pin_d2 = CYD_2432S022C_LCD_PIN_D2;
            cfg.pin_d3 = CYD_2432S022C_LCD_PIN_D3;
            cfg.pin_d4 = CYD_2432S022C_LCD_PIN_D4;
            cfg.pin_d5 = CYD_2432S022C_LCD_PIN_D5;
            cfg.pin_d6 = CYD_2432S022C_LCD_PIN_D6;
            cfg.pin_d7 = CYD_2432S022C_LCD_PIN_D7;
            _bus_instance.config(cfg);
            _panel_instance.setBus(&_bus_instance);
        }

        // Panel configuration
        {
            auto cfg = _panel_instance.config();
            cfg.pin_cs = CYD_2432S022C_LCD_PIN_CS;
            cfg.pin_rst = CYD_2432S022C_LCD_PIN_RST;
            cfg.pin_busy = -1;
            cfg.panel_width = CYD_2432S022C_LCD_HORIZONTAL_RESOLUTION;
            cfg.panel_height = CYD_2432S022C_LCD_VERTICAL_RESOLUTION;
            cfg.offset_x = 0;
            cfg.offset_y = 0;
            cfg.offset_rotation = 0;
            cfg.dummy_read_pixel = 8;
            cfg.dummy_read_bits = 1;
            cfg.readable = true;
            cfg.invert = false;
            cfg.rgb_order = false;  // BGR order for correct colors
            cfg.dlen_16bit = false;
            cfg.bus_shared = true;
            _panel_instance.config(cfg);
        }

        // Backlight configuration
        {
            auto cfg = _light_instance.config();
[... 123 lines cut ...]
    std::string getDescription() const override { return "LovyanGFX-based display for CYD-2432S022C board"; }

private:
    std::unique_ptr<Configuration> configuration;
    LGFX_CYD_2432S022C lcd;
    lv_display_t* lvglDisplay = nullptr;
    bool isStarted = false;
};

std::shared_ptr<tt::hal::display::DisplayDevice> createDisplay() {
    auto touch = createCST820Touch();
    if (!touch) {
        ESP_LOGE(TAG, "Failed to create touch device!");
        return nullptr;
    }

    auto config = std::make_unique<LovyanGFXDisplay::Configuration>(
        CYD_2432S022C_LCD_HORIZONTAL_RESOLUTION,
        CYD_2432S022C_LCD_VERTICAL_RESOLUTION,
        touch
    );
    return std::make_shared<LovyanGFXDisplay>(std::move(config));
}
```
Could y'all help with this?

NellowTCS, 2025-03-23:
@KenVanHoeylandt  Hey there! Would you mind looking through my implementation as I feel like some things are definitely a bit overcomplicated and/or need improvement
It's currently booting (yay) but the rotation settings in the Display Settings make the screen glitch out a bit (but this is fixed by a reboot). Also touch calibration is _so_ difficult, but I'm close!

NellowTCS, 2025-03-23:
(ignore that nothing else compiles, I'm just trying to make an app to test calibration)

NellowTCS, 2025-03-23:
> [@KenVanHoeylandt](https://github.com/KenVanHoeylandt) Would you mind looking through my implementation as I feel like some things are definitely a bit overcomplicated and/or need improvement It's currently booting (yay) but the rotation settings in the Display Settings make the screen glitch out a bit (but this is fixed by a reboot). Also touch calibration is _so_ difficult, but I'm close!

How do you rotate the display without rebooting the entire system?

Shadowtrance, 2025-03-23:
Re: touch calibration... I'm confused. You have CST820Touch in your board folder, capacitive touch doesn't need calibration (as far as I'm aware). That and there's already a CST820 driver in tactility to use (CST816S).
Resistive touch needs calibration (XPT2046...don't remember the last time i used resistive touch but anyway)...

And this is why i said start with a basic "lvgl boots hello world" project to get it going *correctly* with esp-idf drivers and lvgl_port instead of hacking in arduino drivers.

That's just my thoughts anyway. you do you :)

NellowTCS, 2025-03-23:
> Re: touch calibration... I'm confused. You have CST820Touch in your board folder, capacitive touch doesn't need calibration (as far as I'm aware). That and there's already a CST820 driver in [...]

Oh, I should have specified better, it's not really calibration, it's mean to be something to help me know how much the offsets should be for each orientation. 
Also, the reason I basically did the driver from scratch (not a good idea, do not recommend lol) is that I just could not get the CST816S driver to initialize properly with LovyanGFX at all. I'll try again however, as I have an idea as to why it didn't work before! 

And yeah fair point. 
I did do some testing without Tactility and just simple LVGL tests, but with Tactility, using the built in driver (ST7789) didn't work properly with my display (I tried a lot). I used LovyanGFX as the documentation for the board used it and that was the best source apart from a few configurations I found online.

Fortunately, I have everything working except for the Rotation and Touch though, and I'll try the CST816S driver once again!

NellowTCS, 2025-03-24:
Apparently the problem was with my old display as now touch works! (_sigh_)

NellowTCS, 2025-03-24:
I got Rotation and Touch working!
I'll open a pull request now.
Thanks for the help!

closed 2025-03-24 by NellowTCS