TactilityProject/Tactility #202: Bug: Can't run simulator on WSL

x3e, 2025-02-02:
Hi! Thank you for this amazing project. I'm looking forward to mess around with it :)

**Describe the bug**
Under Ubuntu 22.04 for WSL, the build went well. However, when trying to run the simulator I don't get any screen. 

**To Reproduce**
Followed the installation instructions:
sudo apt-get install -y build-essential git cmake mesa-utils
git clone https://github.com/ByteWelder/Tactility.git --recurse-submodules
cd Tactility
cmake -S ./ -B buildsim
cmake --build buildsim --target AppSim
cd Data
export LIBGL_ALWAYS_SOFTWARE=1
../buildsim/App/AppSim

**Expected behavior**
I'm guessing some kind of window should pop up showing the simulated screen.

**Logs**
~/Tactility/Data$ ../buildsim/App/AppSim
[WARN] pthread_attr_setstack failed with return value: 22. Default stack will be used.
[WARN] Increase the stack size to PTHREAD_STACK_MIN.
[WARN] pthread_attr_setstack failed with return value: 22. Default stack will be used.
[WARN] Increase the stack size to PTHREAD_STACK_MIN.
I (0) freertos: starting app_main()
D (0) tactility: run
I (0) tactility: Tactility v0.2.0 on Simulator (simulator)
I (0) system_event: SystemEvent::BootInitHalBegin
I (0) system_event: SystemEvent::BootInitI2cBegin
I (0) i2c: Init
I (0) system_event: SystemEvent::BootInitI2cEnd
I (0) hal: Init power
I (0) lvgl_task: lvgl task starting
I (0) lvgl_task: lvgl task started
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: failed to open /dev/dri/card0: Permission denied

I (32) hal: Mounting sdcard
I (32) system_event: SystemEvent::BootInitHalEnd
I (32) tactility: Registering and starting system services
I (32) service_registry: Adding sdcard
I (32) service_registry: Starting sdcard
I (32) service_registry: Started sdcard
I (32) service_registry: Adding Wifi
I (32) service_registry: Starting Wifi
I (32) service_registry: Started Wifi
I (32) lvglinit: Starting
I (32) system_event: SystemEvent::BootInitLvglBegin
I (32) lvglinit: Touch init
I (32) lvglinit: Keyboard init
I (32) lvglinit: Keyboard started
I (32) lvglinit: Finished
I (32) system_event: SystemEvent::BootInitLvglEnd
I (32) tactility: Registering and starting system services
I (32) service_registry: Adding Loader
I (32) service_registry: Starting Loader
[WARN] pthread_attr_setstack failed with return value: 22. Default stack will be used.
[WARN] Increase the stack size to PTHREAD_STACK_MIN.
I (32) service_registry: Started Loader
I (32) service_registry: Adding Gui
I (32) service_registry: Starting Gui
[WARN] pthread_attr_setstack failed with return value: 22. Default stack will be used.
[WARN] Increase the stack size to PTHREAD_STACK_MIN.
I (33) service_registry: Started Gui
I (33) service_registry: Adding Statusbar
I (33) service_registry: Starting Statusbar
I (33) statusbar: id 0: added
I (33) statusbar: event
I (33) statusbar: id 1: added
I (33) statusbar: event
I (33) statusbar: id 2: added
I (33) statusbar: event
I (33) statusbar: id 0: set visibility 1
I (33) statusbar: event
I (33) statusbar: id 0: set image A:/system/service/Statusbar/wifi_signal_strong_white.png
I (33) statusbar: event
I (33) statusbar: id 0: set visibility 1
I (33) statusbar: event
I (33) statusbar: id 1: set image A:/system/service/Statusbar/sdcard.png
I (33) statusbar: event
I (33) statusbar: id 1: set visibility 1
I (33) statusbar: event
I (33) statusbar: id 2: set image A:/system/service/Statusbar/power_100.png
I (33) statusbar: event
I (33) statusbar: id 2: set visibility 1
I (33) statusbar: event
I (33) service_registry: Started Statusbar
I (33) service_registry: Adding Screenshot
I (33) service_registry: Starting Screenshot
I (33) service_registry: Started Screenshot
I (33) tactility: Registering default apps
I (33) app: Registering manifest AlertDialog
I (33) app: Registering manifest AppList
I (33) app: Registering manifest Boot
I (33) app: Registering manifest Display
I (33) app: Registering manifest Files
I (33) app: Registering manifest Gpio
I (33) app: Registering manifest I2cScanner
I (33) app: Registering manifest I2cSettings
I (33) app: Registering manifest ImageViewer
I (33) app: Registering manifest InputDialog
I (33) app: Registering manifest Launcher
I (33) app: Registering manifest Log
I (33) app: Registering manifest Settings
I (33) app: Registering manifest SelectionDialog
I (33) app: Registering manifest SystemInfo
I (33) app: Registering manifest TextViewer
I (33) app: Registering manifest TimeDateSettings
I (33) app: Registering manifest TimeZone
I (33) app: Registering manifest UsbSettings
I (33) app: Registering manifest WifiApSettings
I (33) app: Registering manifest WifiConnect
I (33) app: Registering manifest WifiManage
I (33) app: Registering manifest Screenshot
I (33) app: Registering manifest Power
I (33) tactility: Registering and starting user services
I (33) tactility: Registering user apps
I (33) app: Registering manifest HelloWorld
I (33) tactility: init starting desktop app
I (33) loader: Start app Boot
I (34) tactility: init complete
I (34) tactility: Processing main dispatcher
I (34) loader: Start by id Boot
I (34) loader: Start with manifest Boot
I (34) loader: App "Boot" state: initial -> initial
I (34) loader: App "Boot" state: initial -> started
I (34) loader: App "Boot" state: started -> showing
I (34) boot: A:/system/app/Boot/logo.png
[WARN] pthread_attr_setstack failed with return value: 22. Default stack will be used.
[WARN] Increase the stack size to PTHREAD_STACK_MIN.
I (34) system_event: SystemEvent::BootSplash
I (34) loader: Stop app
I (34) loader: Start app Launcher
I (34) loader: App "Boot" state: showing -> hiding
I (34) loader: App "Boot" state: hiding -> stopped
I (34) loader: Start by id Launcher
I (34) loader: Start with manifest Launcher
I (34) loader: App "Launcher" state: initial -> initial
I (34) loader: App "Launcher" state: initial -> started
I (34) loader: App "Launcher" state: started -> showing
I (84) statusbar: Update in 54 s
I (84) statusbar: event

**Details (please complete the following information):**
 - ESP type: -
 - Device: Windows 11 + WSL + Ubuntu 22.04
 - Tactility version: 0.2.0 (Main currently won't build on my system)

KenVanHoeylandt, 2025-02-02:
## Building

First things first! It's important that we get the `main` branch working first, because then we are using the same builds to compare with eachother.

If this is the error you got with the `main` branch...
[image]
... the easiest solution I know is to do a new `git checkout` for the project.
If you don't want to do an entirely new checkout, you can try these commands (I haven't tried them yet):
```sh
git submodule update --init --remote --recursive
git submodule update --recursive
```

## Window Issue

### 1. Your logs

```
export LIBGL_ALWAYS_SOFTWARE=1
```
```
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: failed to open /dev/dri/card0: Permission denied
```

It seems that you're trying to force software rendering, but that's is apparently ignored.
SDL then tries to access the graphics card, but the permission is denied.

Running the application with `sudo` is probably going to fix that, but that's not really a solution.

### 2. SDL build features

You could verify that SDL is being built correctly. I had the same window issue before, when I did not have a proper SDL build.

When you do a clean build (`rm -rf buildsim` before building), you should see the X11 and Wayland features on:
[image]
Is this your case?

### 3. Do you have GUI applications enabled in the WSL Settings?

[image]

For reference, a screenshot with my log.

x3e, 2025-02-03:
Oops! I didn't know about that GUI applications setting. After enabling that, I am able to run the pre-built versions. Awesome!

I was not sure how stable the main branch is supposed to be, that's why I didn't focus too much on the build issue.
For the record, I was trying to build the branch main, commit 5ec96f60f19fd771c106b26fe1323c29a9ca19d4.

```
[...lots of uninteresting stuff. No warnings or errors before this...]
[ 95%] Building CXX object Tactility/CMakeFiles/Tactility.dir/Source/app/gpio/Gpio.cpp.o
/home/x3e/Tactility/Tactility/Source/app/gpio/Gpio.cpp: In member function ‘void tt::app::gpio::GpioApp::updatePinWidgets()’:
/home/x3e/Tactility/Tactility/Source/app/gpio/Gpio.cpp:67:17: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   67 |             if ((void*)level != label_user_data) {
      |                 ^~~~~~~~~~~~
/home/x3e/Tactility/Tactility/Source/app/gpio/Gpio.cpp:68:45: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   68 |                 lv_obj_set_user_data(label, (void*)level);
      |                                             ^~~~~~~~~~~~
[ 95%] Building CXX object Tactility/CMakeFiles/Tactility.dir/Source/app/gpio/GpioHal.cpp.o
[ 95%] Building CXX object Tactility/CMakeFiles/Tactility.dir/Source/app/i2cscanner/I2cHelpers.cpp.o
[ 95%] Building CXX object Tactility/CMakeFiles/Tactility.dir/Source/app/i2cscanner/I2cScanner.cpp.o
/home/x3e/Tactility/Tactility/Source/app/i2cscanner/I2cScanner.cpp:16:10: fatal error: format: No such file or directory
   16 | #include <format>
      |          ^~~~~~~~
compilation terminated.
gmake[3]: *** [Tactility/CMakeFiles/Tactility.dir/build.make:370: Tactility/CMakeFiles/Tactility.dir/Source/app/i2cscanner/I2cScanner.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:1049: Tactility/CMakeFiles/Tactility.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:7816: App/CMakeFiles/AppSim.dir/rule] Error 2
gmake: *** [Makefile:2873: AppSim] Error 2
```

Looks like you're using features that have not been enabled / included in my compiler. I have gcc version 11.4.0 installed, which seems to be the default for ubuntu 22.04.

If I have some time (maybe later this week) I can try to upgrade.

Nevertheless, thanks again for your help on the WSL settings!

KenVanHoeylandt, 2025-02-03:
Glad to hear it is fixed! Thanks for giving my project a try and reporting your findings.
Regarding the compile issue: You'll need C++20, so that'll likely require Ubuntu 24.04
I'll close this GitHub issue as the original question is answered/resolved, but feel free to open new ones if you have more questions.

closed 2025-02-03 by KenVanHoeylandt