TactilityProject/Tactility #223: Guition JC3248W535 320*480

bensuffolk, 2025-02-16:
I am looking at this [Guition JC3248W535](https://www.surenoo.com/collections/258657/products/23285361) module for a project. 

I expect I'd want to use it in landscape so 480 x 320 display. I note from your documentation that you say "If the resolution is higher (e.g. 480x240 or larger) then performance tends to degrade."

I wondered how much, and if you thought it was worth investing time setting up a new board for this device, or if I'd be better off just doing some custom LVGL code for my project?

KenVanHoeylandt, 2025-02-16:
The unPhone has the same amount of pixels and the performance degradation isn't making the device unusable. You mainly notice the frame rate going down when scrolling content.

How much it's slower really depends on the driver (and if it's able to use QSPI), but you roughly have twice as many pixels as a T-Deck.
Even on T-Deck, the FPS drops a bunch when scrolling content. In an idle loop, it does about 90fps, but while scrolling it's maybe 10-30 fps depending on the content. (text&widgets are no issue, but icons stored in PSRAM are problematic)

Even if you go with custom LVGL code, I don't expect to get much more performance out of it as opposed to running it with Tactility. Perhaps if you find and use a different driver, but you'll likely end up using the same `esp_lcd` driver anyway:

Having the [esp_lcd driver available](https://components.espressif.com/components/espressif/esp_lcd_axs15231b/versions/1.0.0) is very good news. I don't know what the state is of this particular driver, but it's coming from Espressif itself, so that generally means it works well.

AXS15231B seems to have both 565 and 888 pixel formats, so that's good news: in case that byte swapping is necessary (that's slow due to it being a software operation), you can increase performance a bunch by using 24 bit pixels. At the cost of some memory of course, due to larger buffers.

bensuffolk, 2025-02-16:
Yes this board is wired for qspi to the screen driver, and an interrupt pin on the touch sensor (so I’m assuming that means no polling needed).

I’m not actually expecting much scrolling to be needed on my project so this all pretty promising. 

I have ordered a board and let’s see how it plays out.

Shadowtrance, 2025-02-17:
Yeah this display is a mega pain in the butt tbh. (A nice display but a pain)

Seems the only options are using the included (in the demo zip) available esp-idf driver/lvgl port (which is limited to lvgl 8.x) or Arduino_GFX (don't know what black magic is happening there but it works).

The included demo lv_port has been modified compared to the original to suit this screen in particular it seems along with the specific bsp config. The espressif driver alone isn't enough to get it going with lvgl properly.

I've been banging my head against a wall with it trying to get it going with esp-idf to mixed results.

It seems to NEED the tear config in the bsp files along with i think some changes in the lv_port files as well to actually work, full rotation and all. Again, with lvgl 8.x. :(

Hopefully you have better progress than i have. :)
And i hope you share if you do. :)

stevenliuit, 2025-02-18:
[image]
this board img

stevenliuit, 2025-02-18:
I would like to connect a microphone, is this possible?

bensuffolk, 2025-02-18:
What LVGL is Tactility using? I’m going to want a colour picker which I believe does not exist anymore with version 9?

KenVanHoeylandt, 2025-02-18:
> I would like to connect a microphone, is this possible?

Yes. Any of those free IO pins should be fine as long as they're not already used by the board itself (check the datasheet/docs)

bensuffolk, 2025-02-23:
The old colour picker isn’t great anyway, so I’m happy to look at a work around. Hoping my board will arrive shortly and I can have a go at getting it to work

stevenliuit, 2025-02-26:
> > I would like to connect a microphone, is this possible?
> 
> Yes. Any of those free IO pins should be fine as long as they're not already used by the board itself (check the datasheet/docs)

thx  
I want to use https://github.com/stevenliuit/xiaozhi-esp32 to compile the voice dialogue with deepseek model for JC3248W535, but I found that there is a problem with the screen driver and there is no microphone. Now I don’t know what to do. I added the service to my submission record and it can run and the wifi can be used. Can anyone help me take a look?

KenVanHoeylandt, 2025-02-26:
@stevenliuit Let's keep things on-topic.
Questions `xiaozhi-esp32` can be discussed at https://github.com/78/xiaozhi-esp32/issues

stevenliuit, 2025-03-01:
> [@stevenliuit](https://github.com/stevenliuit) Let's stay on-topic. Questions `xiaozhi-esp32` can be discussed at https://github.com/78/xiaozhi-esp32/issues

thx

stevenliuit, 2025-03-14:
Has there been any progress on this project since then?

NellowTCS, 2025-09-20:
> Has there been any progress on this project since then?

same question.

just curious, don't have this board

NellowTCS, 2025-10-16:
> Pure esp-idf with lvgl_port etc above v8 just won't work correctly due to multiple things, one being the TE pin (something to do with tearing control) which isn't supported in lvgl_port except for rgb / mipi dsi displays (and this one but only with the vendor supplied lvgl_port v8) and [esp-arduino-libs/ESP32_Display_Panel#164 (comment)](https://github.com/esp-arduino-libs/ESP32_Display_Panel/issues/164#issuecomment-2668006767) So it's kinda special...and not really in the good way, just annoying special. lol

I got a build using the TE emulation!
https://github.com/NellowTCS/Tactility/actions/runs/18569790808/artifacts/4291561186

cracktorio, 2025-11-29:
> I got a build using the TE emulation! https://github.com/NellowTCS/Tactility/actions/runs/18569790808/artifacts/4291561186

Hey, that link returns a 404. Is there a way i can compile a working build myself?

Shadowtrance, 2025-11-30:
There is no "working" build currently due to this display controller being a little special and as mentioned previously, It either needs proper TE handling (as is done in lvgl_port/bsp in the [vendor demo](https://github.com/Shadowtrance/jc3248w535c/tree/main/1-Demo/Demo_Arduino/DEMO_LVGL) one of the 2 or a combo of both... i dunno....which is only lvgl 8.x) or some sort of canvas (i guess psram canvas) without TE as Arduino_GFX does... which I'm not sure how it works exactly tbh.

So yeah it's just sitting on the forever backburner atm until something changes, either lvgl_port lets non-rgb/mipi dsi displays handle tearing control or someone else figures out another way.

https://github.com/Shadowtrance/Tactility/tree/cyd-jc3248w535c/Boards/cyd-jc3248w535c is the correct link to my wip (not updated to latest changes)

But if I'm missing something and someone wants to figure it out go nuts because I'm over fighting with this screen all the time, It's even somewhat broken in the latest Arduino_GFX library versions lol

Shadowtrance, 2026-01-31:
For those interested. This device is now as complete as it probably will ever be and works reasonably well.
It has been quite the journey from.. "screw this, in the drawer you go" to "works pretty well and supports DisplayDriver too and full rotation".
Go install it! 😄

KenVanHoeylandt, 2026-01-31:
Thank you very much for persevering with this, @Shadowtrance !
I'll close the issue as it's resolved.