TactilityProject/Tactility #251: How to use a sdCard via SDMMC interface?

water2891, 2025-03-17:
I've made a custom board using a sdCard via SDMMC interface, does the Tactility support it?

KenVanHoeylandt, 2025-03-17:
What do you mean by "support" exactly? In theory, Tactility supports any kind of SDMMC hardware as long as it can be built on top of ESP-IDF. Whether it's supported out-of-the-box depends on how yours is built.

Tactility requires an implementation of `SdCardDevice` (see `SdCardDevice.h`) which practically only needs an implementation of `bool mount(string path)`, `bool unmount()` and `State getState()`.

There is a default driver available that works via a SPI interface (`SpiSdCardDevice`). This driver only requires a configuration as parameters (pins, speed, etc.).

water2891, 2025-03-18:
Well, thank you for your guidance. I am a newbie, I mean how to add an sdCard that uses an SDMMC interface instead of an SPI interface. I wrote SDMMCSdCardDevice with reference to SpiSdCardDevice and mount it with esp_vfs_fat_sdmmc_mount, but I don't know how to implement the getLock method and feel like I don't need it.

water2891, 2025-03-18:
Problem solved, thank you

closed 2025-03-18 by water2891

KenVanHoeylandt, 2025-03-18:
Glad to hear you got it resolved!

The lock is there so shared access to the SD MMC card is not causing troubles. This is only an issue when using SPI where the SD card and screen are sharing a SPI bus. In this case you can just return your own lock instance (e.g. `Mutex`) that you store in your class.