TactilityProject/Tactility #255: Bug:

dm-vev, 2025-03-23:
### **Bug Report: GT911 Touch Controller Initialization Failure on LilyGo T-Deck**  

#### **Describe the bug**  
When flashing the latest firmware build from GitHub Actions onto the LilyGo T-Deck, the system fails to initialize the GT911 touch controller. The following errors appear at startup:  

```
W (1319) GT911: Unable to initialize the I2C address
E (1324) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(140): i2c transaction failed
E (1331) GT911: touch_gt911_read_cfg(389): GT911 read error!
E (1336) GT911: esp_lcd_touch_new_i2c_gt911(159): GT911 init failed
E (1343) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
E (1350) GT911: Driver init failed
```  

Additionally, an assertion failure occurs in `tlsf_free` during memory deallocation:  

```
assert failed: tlsf_free tlsf.c:629 (!block_is_free(block) && "block already marked as free")
```

Backtrace:  

```
0x403762c5:0x3fcb74c0 0x40380f15:0x3fcb74e0 0x4038abd9:0x3fcb7500  
0x403886fa:0x3fcb7620 0x40388598:0x3fcb7640 0x40376c76:0x3fcb7660  
0x4038ac09:0x3fcb7680 0x4208133a:0x3fcb76a0 0x420841c6:0x3fcb76c0  
0x42012651:0x3fcb76e0 0x4201276f:0x3fcb7700 0x420240fa:0x3fcb7770  
0x42024359:0x3fcb7790 0x420129ed:0x3fcb77e0 0x4200fd21:0x3fcb7810  
0x42144f97:0x3fcb7840 0x40381a6e:0x3fcb7870
```

---

#### **To Reproduce**  
Steps to reproduce the behavior:  
1. Flash the latest firmware from GitHub Actions onto the LilyGo T-Deck.  
2. Power on the device.  
3. Observe the boot logs.  
4. The GT911 initialization errors and assertion failure in `tlsf_free` appear.  

---

#### **Expected behavior**  
The GT911 touch controller should initialize successfully without errors, and there should be no memory management assertion failures.  

---

#### **Screenshots**  
[image]

---

#### **Details (please complete the following information):**  
- **ESP type**: ESP32-S3  
- **Device**: LilyGo T-Deck  
- **Tactility version**: Latest GitHub Actions build  

---

#### **Additional context**  
- The issue occurs immediately on boot. (Black screen and reboot cycle)

KenVanHoeylandt, 2025-03-23:
Thanks for the report.
I think this is a combination of 2 issues:
 
1. T-Deck sometimes fails to initialize the GT911 driver. This only happens on the T-Deck and not on the T-Deck plus, so I think that it's a power issue. I've seen it on other GT911 boards too. I've documented it on the [website](https://tactility.one/#/devices/lilygo-tdeck). Try replugging the USB power either after a few seconds, or quicker. It should eventually boot.
2. Touch drivers crash when they are partially started and then try to clean themselves up. I found that this bug is happening on many drivers. I will fix this.

KenVanHoeylandt, 2025-03-23:
The cleanup problem was a bug in the GT911 from esp-bsp:
https://github.com/espressif/esp-bsp/commit/2f17f439f5fc07bcc2a0644bf7844bd0589f72b3#diff-64066628e60d4b6a68ec122e266b701ac6f12dd83535e05cbe440bf0fe9bc071

It's fixed, but the fix is not integrated in Tactility yet.

KenVanHoeylandt, 2025-03-23:
A fix for the crash was merged. When the the touch driver fails to init, the device should now boot, but it won't have working touch.

dm-vev, 2025-03-26:
Hi! Thanks for the previous fix — the device now boots properly without crashing.
However, the GT911 touchscreen still doesn't work on the T-Deck Plus.
Is there any update or workaround planned to get touch functionality working?

KenVanHoeylandt, 2025-03-26:
You're welcome. Touch works fine on my T-Deck and T-Deck Plus (I just tried both), and also on a T-Deck of another user.

Could you do run `idf.py erase-flash` and then flash again?

If the latter doesn't work: does touch work with another firmware? If not, then I'm suspecting hardware failure.

dm-vev, 2025-03-27:
Sometimes, after several power cycles, the touch starts working, but most of the time it fails with the same GT911 init error messages. I’ve tested other firmware (Factory test from LilyGo repository), and touch works consistently there, so this seems to be specific to the current Tactility firmware.
Let me know if there’s anything I can try or test to help diagnose this further.

Thanks again!

dm-vev, 2025-03-27:
I’m attaching screenshots of logs initialization attempts — maybe that helps issue.

Would appreciate any suggestions or if there’s a chance to tweak the GT911.

[image]

[image]

KenVanHoeylandt, 2025-03-27:
So did you try `idf.py erase-flash` before flashing?

dm-vev, 2025-03-28:
Yes

dm-vev, 2025-03-30:
Sure

KenVanHoeylandt, 2025-03-31:
I've identified the problem. A fix will be coming up soon.

KenVanHoeylandt, 2025-04-01:
I'll close the issue. Feel free to re-open it or comment if you still have that issue with the latest Tactility version.

closed 2025-04-01 by KenVanHoeylandt