TactilityProject/Tactility #258: Bug: Unable to build simulator

kicomoco, 2025-03-27:
**Describe the bug**
When running the first cmake command App/CMakeLists.txt is unable to find ../Buildscripts/board.cmake, a suspected pathing issue.....if I edit the file to test and change to not go up a directory it gets past that stage but fails later as it needs the parent directory in the later build stage.

**To Reproduce**
Steps to reproduce the behavior:
From a blank Ubuntu 24.04.2 install
sudo apt-get update
sudo apt-get install git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0 build-essential mesa-utils
mkdir -p ~/esp
cd ~/esp
git clone -b v5.4 --recursive https://github.com/espressif/esp-idf.git
cd ~/esp/esp-idf
./install.sh all
. ./export.sh
cd ~
git clone https://github.com/ByteWelder/Tactility.git --recurse-submodules
cd Tactility
cp sdkconfig.board.m5stack-core2 sdkconfig
cmake -S ./ -B buildsim

**Expected behavior**
Build to complete so can proceed to next step

**Screenshots**
[image]

**Details (please complete the following information):**
 - ESP type: Simulator
 - Device: Simulator
 - Tactility version: main branch

**Additional context**
Full build output (done with pipe redirect to a file because its running in a VM that I don't appear to be able to copy/paste from):

                      [1;35m@@
                     [1;35m@@@
                     [1;35m@@@
                     [1;35m@@@
                     [1;35m@@@
     [36m@@@@@@@@@@@@@@@@[1;35m@@@
    [36m@@@@@@@@@@@@@@@@@[1;35m@@@
              [36m@@@    [1;35m@@@                 [1;37mTactility 0.3.0
              [36m@@@    [1;35m@@@                 [37m   @ ESP-IDF
              [36m@@@[1;35m@@@@@@@@@@@@@@@@@
              [36m@@@[1;35m@@@@@@@@@@@@@@@@
              [36m@@@
              [36m@@@
              [36m@@@
              [36m@@@
              [36m@@

[m
Building with ESP-IDF v5.4
Building board m5stack-core2
Board project: Boards/M5stackCore2
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements are satisfied.
Constraint file: /home/ubuntu/.espressif/espidf.constraints.v5.4.txt
Requirement files:
 - /home/ubuntu/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/ubuntu/.espressif/python_env/idf5.4_py3.12_env/bin/python
NOTICE: Skipping optional dependency: espressif/esp_lcd_st7701
NOTICE: Skipping optional dependency: espressif/esp_tinyusb
NOTICE: Skipping optional dependency: espressif/esp_lcd_st7701
NOTICE: Skipping optional dependency: espressif/esp_tinyusb
NOTICE: Skipping optional dependency: espressif/esp_lcd_st7701
NOTICE: Skipping optional dependency: espressif/esp_tinyusb
NOTICE: Skipping optional dependency: espressif/esp_lcd_st7701
NOTICE: Skipping optional dependency: espressif/esp_tinyusb
NOTICE: Processing 12 dependencies:
NOTICE: [1/12] atanisoft/esp_lcd_ili9488 (1.0.10)
NOTICE: [2/12] atanisoft/esp_lcd_touch_xpt2046 (1.0.5)
NOTICE: [3/12] espressif/cmake_utilities (0.5.3)
NOTICE: [4/12] espressif/esp_io_expander (1.0.1)
NOTICE: [5/12] espressif/esp_io_expander_tca95xx_16bit (1.0.1)
NOTICE: [6/12] espressif/esp_lcd_ili9341 (2.0.0)
NOTICE: [7/12] espressif/esp_lcd_panel_io_additions (1.0.1)
NOTICE: [8/12] espressif/esp_lcd_touch (1.1.2)
NOTICE: [9/12] espressif/esp_lcd_touch_cst816s (1.0.3)
NOTICE: [10/12] espressif/esp_lcd_touch_ft5x06 (1.0.6~1)
NOTICE: [11/12] espressif/esp_lcd_touch_gt911 (1.1.3)
NOTICE: [12/12] idf (5.4.0)
CMake Error at /home/ubuntu/esp/esp-idf/tools/cmake/component.cmake:228 (message):
  CMake Warning (dev) at buildsim/build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 106

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /home/ubuntu/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at App/CMakeLists.txt:5 (include):

    include could not find requested file:

  

      ../Buildscripts/board.cmake

  Call Stack (most recent call first):

    /home/ubuntu/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    /home/ubuntu/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:160 (__component_get_requirements)

  

Call Stack (most recent call first):
  /home/ubuntu/esp/esp-idf/tools/cmake/build.cmake:644 (__component_get_requirements)
  /home/ubuntu/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:55 (project)

-- Configuring incomplete, errors occurred!

KenVanHoeylandt, 2025-03-27:
When you run `. ./export.sh` you apply the ESP-IDF context onto your terminal. When you then run the Tactility build, the Tactility scripting assumes you want to build an actual board instead of a simulator.

The solution is to open a new terminal tab/window and immediately run `cmake -S ./ -B buildsim` from the project root folder.

Thank you for the very detailed reproduction steps!

kicomoco, 2025-03-27:
Ahh excellent thank you! Got a step further...

Do you have any tips of solving issues such as SDL_X11 (Wanted: ON): OFF

I have loads of these and obviously get no window when trying to launch the simulator!

KenVanHoeylandt, 2025-03-27:
It seems that I removed something from the docs that I should put back:

> Note: on Ubuntu 22.04, you need to have export LIBGL_ALWAYS_SOFTWARE=1 in your shell environment.

kicomoco, 2025-03-28:
Hi KenVanHoeylandt,

I'm sorry to keep bugging you...I added the environmental variable and it improved it, but still missing loads of stuff!

```
-- SDL2 was configured with the following options:
--
-- Platform: Linux-6.11.0-21-generic
-- 64-bit: TRUE
-- Compiler: /usr/bin/cc
-- Revision: SDL-prerelease-2.29.2-482-g0efeb82a2
--
-- Subsystems:
-- Atomic: ON
-- Audio: ON
-- Video: ON
-- Render: ON
-- Events: ON
-- Joystick: ON
-- Haptic: ON
-- Hidapi: ON
-- Power: ON
-- Threads: ON
-- Timers: ON
-- File: ON
-- Loadso: ON
-- CPUinfo: ON
-- Filesystem: ON
-- Sensor: ON
-- Locale: ON
-- Misc: ON
--
-- Options:
-- SDL2_DISABLE_INSTALL (Wanted: ON): OFF
-- SDL2_DISABLE_SDL2MAIN (Wanted: OFF): OFF
-- SDL2_DISABLE_UNINSTALL (Wanted: OFF): OFF
-- SDL_3DNOW (Wanted: OFF): OFF
-- SDL_ALSA (Wanted: ON): OFF
-- SDL_ALSA_SHARED (Wanted: ON): OFF
-- SDL_ALTIVEC (Wanted: ON): OFF
-- SDL_ARMNEON (Wanted: OFF): OFF
-- SDL_ARMSIMD (Wanted: OFF): OFF
-- SDL_ARTS (Wanted: ON): OFF
-- SDL_ARTS_SHARED (Wanted: ON): OFF
-- SDL_ASAN (Wanted: OFF): OFF
-- SDL_ASSEMBLY (Wanted: ON): ON
-- SDL_ASSERTIONS (Wanted: auto): auto
-- SDL_BACKGROUNDING_SIGNAL (Wanted: OFF): OFF
-- SDL_CCACHE (Wanted: ON): ON
-- SDL_CLOCK_GETTIME (Wanted: ON): ON
-- SDL_COCOA (Wanted: OFF): OFF
-- SDL_DBUS (Wanted: ON): OFF
-- SDL_DIRECTFB (Wanted: OFF): OFF
-- SDL_DIRECTFB_SHARED (Wanted: OFF): OFF
-- SDL_DIRECTX (Wanted: OFF): OFF
-- SDL_DISKAUDIO (Wanted: ON): ON
-- SDL_DUMMYAUDIO (Wanted: ON): ON
-- SDL_DUMMYVIDEO (Wanted: ON): ON
-- SDL_ESD (Wanted: ON): OFF
-- SDL_ESD_SHARED (Wanted: ON): OFF
-- SDL_FOREGROUNDING_SIGNAL (Wanted: OFF): OFF
-- SDL_FUSIONSOUND (Wanted: OFF): OFF
-- SDL_FUSIONSOUND_SHARED (Wanted: OFF): OFF
-- SDL_GCC_ATOMICS (Wanted: ON): ON
-- SDL_HIDAPI (Wanted: ON): ON
-- SDL_HIDAPI_JOYSTICK (Wanted: ON): ON
-- SDL_HIDAPI_LIBUSB (Wanted: OFF): OFF
-- SDL_IBUS (Wanted: ON): OFF
-- SDL_INSTALL_TESTS (Wanted: OFF): OFF
-- SDL_JACK (Wanted: ON): OFF
-- SDL_JACK_SHARED (Wanted: ON): OFF
-- SDL_KMSDRM (Wanted: ON): OFF
-- SDL_KMSDRM_SHARED (Wanted: ON): OFF
-- SDL_LASX (Wanted: OFF): OFF
-- SDL_LIBC (Wanted: ON): ON
-- SDL_LIBICONV (Wanted: OFF): OFF
-- SDL_LIBSAMPLERATE (Wanted: ON): OFF
-- SDL_LIBSAMPLERATE_SHARED (Wanted: ON): OFF
-- SDL_LIBUDEV (Wanted: ON): OFF
-- SDL_LSX (Wanted: OFF): OFF
-- SDL_METAL (Wanted: OFF): OFF
-- SDL_MMX (Wanted: ON): ON
-- SDL_NAS (Wanted: ON): OFF
-- SDL_NAS_SHARED (Wanted: ON): OFF
-- SDL_OFFSCREEN (Wanted: ON): ON
-- SDL_OPENGL (Wanted: ON): OFF
-- SDL_OPENGLES (Wanted: ON): ON
-- SDL_OSS (Wanted: ON): ON
-- SDL_PIPEWIRE (Wanted: ON): OFF
-- SDL_PIPEWIRE_SHARED (Wanted: ON): OFF
-- SDL_PTHREADS (Wanted: ON): ON
-- SDL_PTHREADS_SEM (Wanted: ON): ON
-- SDL_PULSEAUDIO (Wanted: ON): OFF
-- SDL_PULSEAUDIO_SHARED (Wanted: ON): OFF
-- SDL_RENDER_D3D (Wanted: OFF): OFF
-- SDL_RENDER_METAL (Wanted: OFF): OFF
-- SDL_RPATH (Wanted: ON): ON
-- SDL_RPI (Wanted: ON): OFF
-- SDL_SNDIO (Wanted: ON): OFF
-- SDL_SNDIO_SHARED (Wanted: ON): OFF
-- SDL_SSE (Wanted: ON): ON
-- SDL_SSE2 (Wanted: ON): ON
-- SDL_SSE3 (Wanted: ON): ON
-- SDL_SSEMATH (Wanted: ON): ON
-- SDL_STATIC_PIC (Wanted: OFF): OFF
-- SDL_SYSTEM_ICONV (Wanted: ON): ON
-- SDL_TESTS (Wanted: OFF): OFF
-- SDL_VENDOR_INFO (Wanted: ): OFF
-- SDL_VIRTUAL_JOYSTICK (Wanted: ON): ON
-- SDL_VIVANTE (Wanted: ON): OFF
-- SDL_VULKAN (Wanted: ON): ON
-- SDL_WASAPI (Wanted: OFF): OFF
-- SDL_WAYLAND (Wanted: ON): OFF
-- SDL_WAYLAND_LIBDECOR (Wanted: ON): OFF
-- SDL_WAYLAND_LIBDECOR_SHARED (Wanted: ON): OFF
-- SDL_WAYLAND_QT_TOUCH (Wanted: ON): OFF
-- SDL_WAYLAND_SHARED (Wanted: ON): OFF
-- SDL_X11 (Wanted: ON): OFF
-- SDL_X11_SHARED (Wanted: ON): OFF
-- SDL_X11_XCURSOR (Wanted: ON): OFF
-- SDL_X11_XDBE (Wanted: ON): OFF
-- SDL_X11_XFIXES (Wanted: ON): OFF
-- SDL_X11_XINPUT (Wanted: ON): OFF
-- SDL_X11_XRANDR (Wanted: ON): OFF
-- SDL_X11_XSCRNSAVER (Wanted: ON): OFF
-- SDL_X11_XSHAPE (Wanted: ON): OFF
-- SDL_XINPUT (Wanted: OFF): OFF
```

I'm trying to get the simulator working so I can write some apps....I would love an SSH client on it!

On a side note....I'm also working on getting Tactility to work on a ClockworkPi PicoCalc with a Waveshare ESP32S3 Pico core...got the screen working and the keyboard is sort of working......I can see from the debug it is now registering keypresses but I haven't worked out how to navigate the UI without a touchscreen so can't actually interact with Tactility lol

kicomoco, 2025-03-28:
Well I made some progress on my own!

For anybody else who finds this I needed to install the sdl2 development libraries
```
sudo apt-get install libsdl2-dev
```

I now have the simulator running!

KenVanHoeylandt, 2025-03-28:
Glad to hear it works! Thanks for figuring out the solution - I'll update the documentation right away
I'm guessing that `libsdl2-dev` comes with some dependencies that fix the issue.

closed 2025-03-28 by KenVanHoeylandt