TactilityProject/Tactility #265: Bug: Waveshare S3-Touch-4.3 not working

mattstep, 2025-03-30:
**Describe the bug**
I built and flashed my waveshare s3-lcd-4.3 with touch (regular, not the B model), and I get a white screen and the uart is reporting a warning continually "W (581124) wifi_service: Auto-connect: radio state not ok (5, 0, 0)"

**To Reproduce**
Steps to reproduce the behavior:
1. At commit 3f1bfee3f5812de78c1b243fef776547b334abf7
2. `cp sdkconfig.board.waveshare-s3-touch-43 sdkconfig`
3. `idf.py build`
4. `idf.py -p /dev/ttyACM0 flash monitor`

**Expected behavior**
Tactility running on my waveshare s3-lcd-4.3

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Details (please complete the following information):**
 - ESP type: ESP32-S3
 - Device: WaveshareS3Touch43
 - Tactility version: Main branch, commit 3f1bfee3f5812de78c1b243fef776547b334abf7

**Additional context**
I can build and run on my t-deck just fine, so my dev environment seems setup correctly.

```idf.py --version
ESP-IDF v5.4
```

mattstep, 2025-03-30:
Just hopped into discord as well as Mattstep, so feel free to ping me there as well. Would love to get this working on my waveshare.

mattstep, 2025-03-30:
I found `W (1255) GT911: Unable to initialize the I2C address` in the logs

Shadowtrance, 2025-03-31:
> I found `W (1255) GT911: Unable to initialize the I2C address` in the logs

That is pretty normal for GT911 for whatever reason. A lot of the time you get that in the log and it's working fine.
I get it on my 4.3 inch (not waveshare) but it's no issue. GT911 is just a pain in the butt tbh.

KenVanHoeylandt, 2025-03-31:
Keep in mind that the main branch is not an official release. I generally only test on 1 device (sometimes more) when I do changes.
The warning was left in from last night, but I'll be removing it in the next PR. It's not problematic as it was just part of debugging.

The white screen is a different problem, though.

That said, I was getting crashes with flashing screen due to a GT911 bug. I've identified the issue and a fix will be coming up very soon.

KenVanHoeylandt, 2025-03-31:
Fixes were pushed to the `main` branch. Can you test again?

Note that there's a known issue where the screen sometimes shifts horizontally when the device does heavy operations like reading files or starting WiFi. It's a problem with all devices that have an RGB/i80 display currently.

mattstep, 2025-04-01:
No, I wasn't able to get it working from master. I'll try from the releases tomorrow, just to compare.

KenVanHoeylandt, 2025-04-01:
Discussed in Discord. 
Fix is implemented and pending: https://github.com/ByteWelder/Tactility/pull/269

KenVanHoeylandt, 2025-04-01:
Fixes were merged.

closed 2025-04-01 by KenVanHoeylandt