TactilityProject/Tactility #267: When I use the KeyboardDevice class to create my own input device (encoder), it causes the soft keyboard to be disabled

water2891, 2025-04-01:
Now I temporarily let the start method of my custom KeyboardDevice class return false to prevent the soft keyboard from being deactivated. I wonder if this is officially recommended?

KenVanHoeylandt, 2025-04-01:
It's not recommended, because the effect is that your device will not be discoverable. 
For now, it's an ok work-around, I guess.
I'll add a keyboard setting app in the coming weeks, so you can still get the software keyboard when you want to. I'll let you know here when it's available.

KenVanHoeylandt, 2025-08-30:
There is now an `EncoderDevice` available.  An example implementation can be found here:
https://github.com/ByteWelder/Tactility/blob/main/Boards/LilygoTLoraPager/Source/hal/TpagerEncoder.cpp
https://github.com/ByteWelder/Tactility/blob/main/Boards/LilygoTLoraPager/Source/hal/TpagerEncoder.h
It can co-exist with a physical keyboard, as I am now able to test such a combination with the LilyGO T-Lora Pager.

closed 2025-08-30 by KenVanHoeylandt