TactilityProject/Tactility #268: Consider adding some UI configuration

water2891, 2025-04-01:
My development board is self-made, it uses a GC9A01 driven round screen. The current UI alignment is not very friendly for circular screens, I tried to modify some of the source code to make it work.

I hope you will consider adding some UI configuration in future versions to make them have a better experience on circular screen. Thank you.

KenVanHoeylandt, 2025-04-01:
Thanks for the suggestion. I'll think about it. I plan to start supporting small screen devices this year (e.g. M5 Cardputer, M5 StickC Plus2). I'll keep this in consideration when I rework the UI for those devices.
 
I'm not sure if I can come up with a solution for round screens at the same time, but I can make sure that the future change fits into the new code patterns. 
(e.g. rendering a switch differently on a small screen, showing a different type of keyboard, rendering the taskbar/toolbars differently, etc)

I'm leaning towards an Android-like approach where we swap out layouts/graphics/etc.

KenVanHoeylandt, 2026-11-08:
Some progress has been made on this aspect. Display drivers now support insets and shape specifications. Some applications are already adapted for these insets, but more needs to be done.