TactilityProject/Tactility #273: Add support of the CYB 2.2 inch capacitive screen. ESP32-2432S022

T-REX-XP, 2025-04-08:
Could you please add support for the CYB ESP32-2432S022 board?

**Documentation and examples:** [link](http://pan.jczn1688.com/1/ESP32%20module)

```
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:15:18:d4:59:d4
```
[image]
[image]

Thanks in advance

NellowTCS, 2025-04-08:
I am working on that, so 👍.
(check #197)

If you want you can try out the latest firmware build at 

[Tactility-cyd-2432s022c.zip](https://github.com/user-attachments/files/19652581/Tactility-cyd-2432s022c.zip)

NellowTCS, 2025-04-08:
> I am working on that, so 👍. (check [#197](https://github.com/ByteWelder/Tactility/issues/197))
> 
> If you want you can try out the latest firmware build at
> 
> [Tactility-cyd-2432s022c.zip](https://github.com/user-attachments/files/19652581/Tactility-cyd-2432s022c.zip)

Just so yk, I am not really sure if it works 😅

T-REX-XP, 2025-04-09:
It doesn't work. Just a white screen.

Flash log:
```
PS D:\Downloads\Tactility-cyd-2432s022c\Tactility-cyd-2432s022c> ./flash.ps1 COM6
esptool.py v4.8.1
Serial port COM6
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...........
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:15:18:d4:59:d4
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.5s
Hard resetting via RTS pin...
esptool.py v4.8.1
Serial port COM6
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:15:18:d4:59:d4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x001f2fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00310000 to 0x0034ffff...
Flash will be erased from 0x00350000 to 0x0038ffff...
Compressed 27616 bytes to 17341...
Wrote 27616 bytes (17341 compressed) at 0x00001000 in 1.6 seconds (effective 134.0 kbit/s)...
Hash of data verified.
Compressed 1977984 bytes to 1164148...
Wrote 1977984 bytes (1164148 compressed) at 0x00010000 in 102.3 seconds (effective 154.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 131...
Wrote 3072 bytes (131 compressed) at 0x00008000 in 0.0 seconds (effective 884.0 kbit/s)...
Hash of data verified.
Compressed 262144 bytes to 19149...
Wrote 262144 bytes (19149 compressed) at 0x00310000 in 2.6 seconds (effective 818.4 kbit/s)...
Hash of data verified.
Compressed 262144 bytes to 543...
Wrote 262144 bytes (543 compressed) at 0x00350000 in 0.8 seconds (effective 2776.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

ESP log:
```
Rebooting...
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6592
load:0x40078000,len:16656
[... 69 lines cut ...]
I (862) hal: Init power
I (864) YellowTouch: Creating YellowTouch
I (868) YellowTouch: YellowTouch created: 0x3ffd0e04
I (872) i80display: Starting
E (875) lcd_panel.io.i80: esp_lcd_new_i80_bus(164): no mem for format buffer
E (882) i80display: Failed to create I80 bus
E (886) YellowDisplay: Failed to initialize i80 display
E (891) CYD2432S022C: Failed to start display during boot
E (896) check: Failed: configuration.initBoot()
E (900) kernel: Task: main
E (902) kernel: Stack watermark: 5504
E (906) kernel: default caps:
E (908) kernel:   total: 281264
E (911) kernel:   free: 189624
E (914) kernel:   min free: 188220
E (917) kernel: internal caps:
E (920) kernel:   total: 325032
E (923) kernel:   free: 232564
E (926) kernel:   min free: 231128
System halted. Connect debugger for more info.

Backtrace: 0x40081c55:0x3ffbee70 0x4008a875:0x3ffbee90 0x400da258:0x3ffbeeb0 0x400f82b8:0x3ffbeed0 0x400e43a4:0x3ffbef30 0x400e42cc:0x3ffbef60 0x400df3be:0x3ffbef90 0x4022e7b6:0x3ffbefc0 0x4008ac6a:0x3ffbeff0
ELF file SHA256: 4a1061fec
Rebooting...
```

T-REX-XP, 2025-04-09:
Here is the M5Stick Launcher for that board -[ https://github.com/bmorcelli/M5Stick-Launcher/blob/main/boards/CYD-2432S028/platformio.ini]( https://github.com/bmorcelli/M5Stick-Launcher/blob/main/boards/CYD-2432S028/platformio.ini).

It may be useful to find the pinout or other configuration for the display.

NellowTCS, 2025-04-09:
> It doesn't work. Just a white screen.
>
> Flash log:
>
> ```
> PS D:\Downloads\Tactility-cyd-2432s022c\Tactility-cyd-2432s022c> ./flash.ps1 COM6
> esptool.py v4.8.1
> Serial port COM6
> Connecting......
> Detecting [...]

This is actually great, b/c I know exactly what the problem is:
Not enough RAM!
I'll work on a fix 👍.

NellowTCS, 2025-04-09:
Try this build out:

[Tactility-cyd-2432s022c.zip](https://github.com/user-attachments/files/19672986/Tactility-cyd-2432s022c.zip)

NellowTCS, 2025-04-09:
> I (872) i80display: Starting
> assert failed: lvgl_port_lock esp_lvgl_port.c:159 (lvgl_port_ctx.lvgl_mux && "lvgl_port_init must be called first")
> 
> Backtrace: 0x40081c55:0x3ffbeaf0 0x4008a875:0x3ffbeb10 0x4008f221:0x3ffbeb30 0x4010c991:0x3ffbec50 0x4010d369:0x3ffbec70 0x400e3d63:0x3ffbeca0 0x400e2b78:0x3ffbed90 0x400e2001:0x3ffbeea0 0x400f82eb:0x3ffbeed0 0x400e4398:0x3ffbef30 0x400e42c0:0x3ffbef60 0x400df3b2:0x3ffbef90 0x4022e802:0x3ffbefc0 0x4008ac6a:0x3ffbeff0
> 
> ELF file SHA256: 2a073ccce
> Rebooting...
> ```

I am working on a fix.

NellowTCS, 2025-04-09:
Here's the latest build:

[Tactility-cyd-2432s022c.zip](https://github.com/user-attachments/files/19675174/Tactility-cyd-2432s022c.zip)

NellowTCS, 2025-04-10:
also this may be due to autocorrect but the title says CYB not CYD.

T-REX-XP, 2025-04-10:
Still broken: 
```
I (1584) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1586) spi_sdcard: Mounting failed (ESP_ERR_TIMEOUT)
E (1591) spi_sdcard: Mount failed for /sdcard
W (1595) hal: SD card mount failed (init can continue)
I (1600) devices: Registered SD Card with id 0
I (1604) system_event: BootInitHalEnd
I (1608) tactility: Registering and starting system services
I (1613) service_registry: Adding Gps
I (1616) service_registry: Starting Gps
I (1620) service_registry: Started Gps
I (1623) service_registry: Adding sdcard
I 1627) service_registry: Starting sdcard
I (1631) service_registry: Started sdcard
I (1634) service_registry: Adding Wifi
I (1638) service_registry: Starting Wifi
I (1642) service_registry: Started Wifi
I (1645) service_registry: Adding EspNow
I (1649) service_registry: Starting EspNow
I (1653) service_registry: Started EspNow
I (1656) lvgl_init: Starting
I (1659) system_event: BootInitLvglBegin
I (1663) YellowTouch: Creating YellowTouch
I (1663) LVGL: Starting LVGL task
I (1667) YellowTouch: YellowTouch created: 0x3ffd83d8
I (1674) YellowDisplay: Buffer size: 15360 bytes
I (1678) i80display: Starting
E (1681) lcd_panel.io.i80: esp_lcd_new_i80_bus(164): no mem for format buffer
E (1688) i80display: Failed to create I80 bus
E (1692) YellowDisplay: Failed to initialize i80 display
E (1697) lvgl_init: Display start failed
I (1701) tactility: Registering and starting system services
I (1706) service_registry: Adding Loader
I (1710) service_registry: Starting Loader
I (1714) service_registry: Started Loader
I (1714) Thread: Starting loader_dispatcher
I (1717) service_registry: Adding Gui
I (1725) service_registry: Starting Gui

assert failed: tt::service::gui::Gui* tt::service::gui::gui_alloc() Gui.cpp:42 (screen_root != nullptr)

Backtrace: 0x40081c55:0x3ffbecc0 0x4008a875:0x3ffbece0 0x4008f221:0x3ffbed00 0x40100d76:0x3ffbee20 0x40100e09:0x3ffbee80 0x400fdcd3:0x3ffbeea0 0x400fdd7c:0x3ffbeee0 0x400fdd9c:0x3ffbef20 0x400e4239:0x3ffbef50 0x400e4394:0x3ffbef70 0x400df38e:0x3ffbefa0 0x4022e8da:0x3ffbefd0 0x4008ac6a:0x3ffbf000

ELF file SHA256: 6ca4e0381
```

NellowTCS, 2025-04-10:
That's pretty confusing, I did set that buffer to 37.5KB, which should be small enough.

T-REX-XP, 2025-04-10:
Still doesn't work. ((

```
I (1659) service_registry: Starting EspNow
I (1663) service_registry: Started EspNow
I (1666) lvgl_init: Starting
I (1669) system_event: BootInitLvglBegin
I (1673) YellowTouch: Creating YellowTouch
I (1673) LVGL: Starting LVGL task
I (1677) YellowTouch: YellowTouch created: 0x3ffd83d8
I (1684) YellowDisplay: DMA heap free: 79412
I (1688) YellowDisplay: Heap free before LVGL init: 79412
I (1693) YellowDisplay: Heap free after LVGL init: 79412
I (1698) YellowDisplay: DMA heap free after LVGL init: 79412
I (1704) YellowDisplay: Buffer size: 15360 bytes
I (1708) i80display: Starting
E (1711) lcd_panel.io.i80: esp_lcd_new_i80_bus(164): no mem for format buffer
E (1717) i80display: Failed to create I80 bus
E (1721) YellowDisplay: Failed to initialize i80 display
E (1727) lvgl_init: Display start failed
I (1730) tactility: Registering and starting system services
I (1736) service_registry: Adding Loader
I (1739) service_registry: Starting Loader
I (1744) service_registry: Started Loader
I (1744) Thread: Starting loader_dispatcher
I (1747) service_registry: Adding Gui
I (1754) service_registry: Starting Gui

assert failed: tt::service::gui::Gui* tt::service::gui::gui_alloc() Gui.cpp:42 (screen_root != nullptr)

Backtrace: 0x40081c55:0x3ffbecc0 0x4008a875:0x3ffbece0 0x4008f221:0x3ffbed00 0x40100dba:0x3ffbee20 0x40100e4d:0x3ffbee80 0x400fdd17:0x3ffbeea0 0x400fddc0:0x3ffbeee0 0x400fdde0:0x3ffbef20 0x400e427d:0x3ffbef50 0x400e43d8:0x3ffbef70 0x400df39e:0x3ffbefa0 0x4022e91e:0x3ffbefd0 0x4008ac6a:0x3ffbf000
ELF file SHA256: f9e96d2d5
Rebooting...
```

NellowTCS, 2025-04-10:
> I (1684) YellowDisplay: DMA heap free: 79412
> I (1688) YellowDisplay: Heap free before LVGL init: 79412
> I (1693) YellowDisplay: Heap free after LVGL init: 79412
> I (1698) YellowDisplay: DMA heap free [...]

That's super weird, and 🤦‍♂️, I put the logs in the wrong place.

NellowTCS, 2025-04-10:
Here is an updated version that should work:

[Tactility-cyd-2432s022c.zip](https://github.com/user-attachments/files/19689702/Tactility-cyd-2432s022c.zip)

T-REX-XP, 2025-04-10:
Still doesn't work
```

I (1633) service_registry: Adding Gps
I (1636) service_registry: Starting Gps
I (1640) service_registry: Started Gps
I (1643) service_registry: Adding sdcard
I (1647) service_registry: Starting sdcard
I (1651) service_registry: Started sdcard
I (1654) service_registry: Adding Wifi
I (1658) service_registry: Starting Wifi
I (1662) service_registry: Started Wifi
I (1665) service_registry: Adding EspNow
I (1669) service_registry: Starting EspNow
I (1673) service_registry: Started EspNow
I (1676) lvgl_init: Starting
I (1679) system_event: BootInitLvglBegin
I (1683) YellowTouch: Creating YellowTouch
I (1683) LVGL: Starting LVGL task
I (1687) YellowTouch: YellowTouch created: 0x3ffd83c0
I (1694) YellowDisplay: DMA heap free: 87628
I (1698) YellowDisplay: Heap free before LVGL init: 87628
I (1703) YellowDisplay: Heap free after LVGL init: 87628
I (1708) YellowDisplay: DMA heap free after LVGL init: 87628
I (1714) YellowDisplay: Buffer size: 15360 bytes
I (1718) i80display: Starting
I (1721) i80display: DMA heap free before bus init: 87448
E (1726) lcd_panel.io.i80: esp_lcd_new_i80_bus(164): no mem for format buffer
E (1733) i80display: Failed to create I80 bus
E (1737) YellowDisplay: Failed to initialize i80 display
E (1742) lvgl_init: Display start failed
I (1745) tactility: Registering and starting system services
I (1751) service_registry: Adding Loader
I (1754) service_registry: Starting Loader
I (1759) service_registry: Started Loader
I (1759) Thread: Starting loader_dispatcher
I (1762) service_registry: Adding Gui
I (1769) service_registry: Starting Gui
assert failed: tt::service::gui::Gui* tt::service::gui::gui_alloc() Gui.cpp:42 (screen_root != nullptr)

Backtrace: 0x40081c55:0x3ffbecc0 0x4008a875:0x3ffbece0 0x4008f221:0x3ffbed00 0x40100dfe:0x3ffbee20 0x40100e91:0x3ffbee80 0x400fdd5b:0x3ffbeea0 0x400fde04:0x3ffbeee0 0x400fde24:0x3ffbef20 0x400e42bd:0x3ffbef50 0x400e4418:0x3ffbef70 0x400df3a6:0x3ffbefa0 0x4022e962:0x3ffbefd0 0x4008ac6a:0x3ffbf000

ELF file SHA256: c4686ffca

Rebooting...
```

NellowTCS, 2025-04-10:
This should be working. It has more than enough RAM.
Any idea @KenVanHoeylandt ?

NellowTCS, 2025-04-10:
@T-REX-XP 
Ok so I'll ask him, but can you try one last time with the below firmware?

https://github.com/NellowTCS/Tactility/actions/runs/14390886400/artifacts/2923426472

T-REX-XP, 2025-04-11:
Still a white screen, but no more crashes.

```
��&io���F���ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:6592

load:0x40078000,len:16656

load:0x40080400,len:4

load:0x40080404,len:4268

entry 0x40080658

I (29) boot: ESP-IDF v5.4 2nd stage bootloader

I (29) boot: compile time Apr 10 2025 21:44:28

I (29) boot: Multicore bootloader

I (31) boot: chip revision: v3.1

I (33) qio_mode: Enabling default flash chip QIO

I (38) boot.esp32: SPI Speed      : 40MHz

I (41) boot.esp32: SPI Mode       : QIO

I (45) boot.esp32: SPI Flash Size : 4MB

I (49) boot: Enabling RNG early entropy source...

I (53) boot: Partition Table:

I (56) boot: ## Label            Usage          Type ST Offset   Length

I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (75) boot:  2 factory          factory app      00 00 00010000 00300000

I (82) boot:  3 system           Unknown data     01 81 00310000 00040000

I (88) boot:  4 data             Unknown data     01 81 00350000 00040000

I (95) boot: End of partition table

I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=64a38h (412216) map

I (224) esp_image: segment 1: paddr=00074a60 vaddr=3ff80000 size=0001ch (    28) load

I (225) esp_image: segment 2: paddr=00074a84 vaddr=3ffb0000 size=05fc8h ( 24520) load

I (236) esp_image: segment 3: paddr=0007aa54 vaddr=40080000 size=055c4h ( 21956) load

I (243) esp_image: segment 4: paddr=00080020 vaddr=400d0020 size=163390h (1454992) map

I (663) esp_image: segment 5: paddr=001e33b8 vaddr=400855c4 size=0ff44h ( 65348) load

I (696) boot: Loaded app from partition at offset 0x10000
[... 244 lines cut ...]
I (2194) app: Registering manifest UsbSettings
I (2198) app: Registering manifest WifiApSettings
I (2202) app: Registering manifest WifiConnect
I (2206) app: Registering manifest WifiManage
I (2210) app: Registering manifest Chat
I (2214) app: Registering manifest CrashDiagnostics
I (2219) tactility: Registering and starting user services
I (2224) tactility: Registering user apps
I (2227) app: Registering manifest Calibration
I (2231) app: Registering manifest TactilityNews
I (2237) app: Registering manifest TactileWeb
I (2240) app: Registering manifest Clock
I (3107) loader: Stop app
I (3107) loader: stopApp()
I (3107) loader: dispatched
I (3108) loader: Start app Launcher
I (3108) loader: App "Boot" state: showing -> hiding
I (3108) Thread: Stopped boot
I (3111) loader: App "Boot" state: hiding -> stopped
I (3118) loader: Free heap: 83612
I (3122) loader: Start by id Launcher
I (3125) loader: App "Launcher" state: initial -> initial
I (3130) loader: App "Launcher" state: initial -> started
I (3135) loader: App "Launcher" state: started -> showing
```

NellowTCS, 2025-04-11:
That's better, but the same issue I had before :/

NellowTCS, 2025-04-19:
So I spent quite a while remaking the driver and fixing a bunch of stuff.
This should work better now:
[Tactility-cyd-2432s022c.zip](https://github.com/NellowTCS/Tactility/actions/runs/14544593699/artifacts/2972541265)

T-REX-XP, 2025-04-21:
> So I spent quite a while remaking the driver and fixing a bunch of stuff. This should work better now: [Tactility-cyd-2432s022c.zip](https://github.com/NellowTCS/Tactility/actions/runs/14544593699/artifacts/2972541265)

Hi. This time the screen is blinking.

```
 (882) YellowDisplay: Heap free before LVGL init: 189296
I (888) LVGL: Starting LVGL task
I (890) YellowDisplay: LVGL port initialized
I (894) YellowDisplay: Heap free after LVGL init: 180232
I (899) YellowDisplay: DMA heap free after LVGL init: 180232
I (905) YellowDisplay: Configured I80Display for 240x320
I (910) i80display: Starting I80 Display
I (914) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1249) i80display: I80 Display initialization completed successfully
I (1249) pwm_backlight: Init
I (1250) YellowDisplay: Display started successfully
W (1252) YellowDisplay: Failed to get backlight duty, using default value
I (1258) YellowDisplay: Backlight duty set to 50
I (1263) YellowDisplay: Backlight duty set to default 50
I (1268) CYD2432S022C: Boot initialization completed successfully
I (1273) i80display: Stopping I80 Display
I (1278) i80display: I80 Display stopped successfully
I (1752) LVGL: Stopped LVGL task
I (1753) YellowDisplay: LVGL port deinitialized
I (1753) YellowDisplay: Display stopped successfully
I (1753) hal: Mounting sdcard
I (1755) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1764) spi_sdcard: Mounting /sdcard
I (1768) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1797) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (1798) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1800) spi_sdcard: Mounting failed (ESP_ERR_TIMEOUT)
E (1805) spi_sdcard: Mount failed for /sdcard
W (1809) hal: SD card mount failed (init can continue)
I (1814) devices: Registered SD Card with id 0
I (1818) system_event: BootInitHalEnd
I (1822) tactility: Registering and starting system services
I (1827) service_registry: Adding Gps
I (1830) service_registry: Starting Gps
I (1834) service_registry: Started Gps
I (1838) service_registry: Adding sdcard
I (1841) service_registry: Starting sdcard
I (1845) service_registry: Started sdcard
I (1848) service_registry: Adding Wifi
I (1852) service_registry: Starting Wifi
I (1856) service_registry: Started Wifi
I (1859) service_registry: Adding EspNow
I (1863) service_registry: Starting EspNow
I (1867) service_registry: Started EspNow
I (1870) lvgl_init: Starting
I (1873) system_event: BootInitLvglBegin
I (1877) YellowTouch: Creating YellowTouch
I (1877) LVGL: Starting LVGL task
I (1881) YellowTouch: YellowTouch created: 0x3ffd2434
I (1888) YellowDisplay: DMA heap free: 173672
I (1892) YellowDisplay: Heap free before LVGL init: 173672
I (1897) YellowDisplay: Heap free after LVGL init: 173672
I (1902) YellowDisplay: DMA heap free after LVGL init: 173672
I (1908) YellowDisplay: Configured I80Display for 240x320
I (1913) i80display: Starting I80 Display
I (1917) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2201) i80display: I80 Display initialization completed successfully
I (2203) pwm_backlight: Init
W (2204) ledc: GPIO 0 is not usable, maybe conflict with others
E (2204) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position
I (2206) YellowDisplay: Display started successfully
E (2216) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position
W (2220) YellowDisplay: Failed to get backlight duty, using default value
E (2229) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position
I (2235) YellowDisplay: Backlight duty set to 50
E (2244) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position
I (2248) YellowDisplay: Backlight duty set to default 50
E (2257) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position
I (2261) YellowDisplay: Backlight duty set to 0
assert failed: std::shared_ptr<tt::hal::display::DisplayDevice> tt::lvgl::initDisplay(const tt::hal::Configuration&) Init.cpp:40 (existing_display_user_data == nullptr)
Backtrace: 0x40081c55:0x3ffbedf0 0x4008a875:0x3ffbee10 0x4008f221:0x3ffbee30 0x400fe5c8:0x3ffbef50 0x400fe632:0x3ffbef80 0x400e4fc5:0x3ffbefd0 0x400df38a:0x3ffbf000 0x40231bf6:0x3ffbf030 0x4008ac6a:0x3ffbf060
ELF file SHA256: 02badc7c7
Rebooting...

```

NellowTCS, 2025-04-21:
> Hi. This time the screen is blinking.

Is it flashing white or multiple colors?

NellowTCS, 2025-04-21:
Maybe try this:
https://github.com/NellowTCS/Tactility/actions/runs/14581532653/artifacts/2981633983

T-REX-XP, 2025-04-22:
> Is it flashing white or multiple colors?

White collor

T-REX-XP, 2025-04-22:
> Maybe try this: https://github.com/NellowTCS/Tactility/actions/runs/14581532653/artifacts/2981633983

```
I (865) system_event: BootInitUartEnd

I (869) hal: Init power

I (871) YellowTouch: Creating YellowTouch

I (875) YellowTouch: YellowTouch created: 0x3ffd0e8c

I (879) YellowDisplay: DMA heap free: 189296

I (883) YellowDisplay: Heap free before LVGL init: 189296

I (889) LVGL: Starting LVGL task

I (891) YellowDisplay: LVGL port initialized

I (895) YellowDisplay: Heap free after LVGL init: 180232

I (900) YellowDisplay: DMA heap free after LVGL init: 180232

I (906) YellowDisplay: Configured I80Display for 240x320

I (911) i80display: Starting I80 Display

I (914) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

I (1247) i80display: I80 Display initialization completed successfully

I (1248) pwm_backlight: Init

I (1249) YellowDisplay: Display started successfully

W (1250) YellowDisplay: Failed to get backlight duty, using default value

I (1256) YellowDisplay: Backlight duty set to 50

I (1261) YellowDisplay: Backlight duty set to default 50

I (1266) CYD2432S022C: Boot initialization completed successfully

I (1272) i80display: Stopping I80 Display

I (1277) i80display: I80 Display stopped successfully

I (1750) LVGL: Stopped LVGL task

I (1751) YellowDisplay: LVGL port deinitialized

I (1751) YellowDisplay: Display stopped successfully

I (1751) hal: Mounting sdcard

I (1753) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

I (1762) spi_sdcard: Mounting /sdcard

I (1766) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

E (1795) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).

I (1796) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

E (1798) spi_sdcard: Mounting failed (ESP_ERR_TIMEOUT)

E (1803) spi_sdcard: Mount failed for /sdcard

W (1807) hal: SD card mount failed (init can continue)

I (1812) devices: Registered SD Card with id 0

I (1816) system_event: BootInitHalEnd

I (1820) tactility: Registering and starting system services

I (1825) service_registry: Adding Gps

I (1828) service_registry: Starting Gps

I (1832) service_registry: Started Gps

I (1835) service_registry: Adding sdcard

I (1839) service_registry: Starting sdcard

I (1843) service_registry: Started sdcard

I (1846) service_registry: Adding Wifi

I (1850) service_registry: Starting Wifi

I (1854) service_registry: Started Wifi

I (1857) service_registry: Adding EspNow

I (1861) service_registry: Starting EspNow

I (1865) service_registry: Started EspNow

I (1868) lvgl_init: Starting

I (1871) system_event: BootInitLvglBegin

I (1876) YellowTouch: Creating YellowTouch

I (1875) LVGL: Starting LVGL task

I (1879) YellowTouch: YellowTouch created: 0x3ffd2434

I (1886) YellowDisplay: DMA heap free: 173672

I (1890) YellowDisplay: Heap free before LVGL init: 173672

I (1895) YellowDisplay: Heap free after LVGL init: 173672

I (1900) YellowDisplay: DMA heap free after LVGL init: 173672

I (1906) YellowDisplay: Configured I80Display for 240x320

I (1911) i80display: Starting I80 Display

I (1915) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

I (2199) i80display: I80 Display initialization completed successfully

I (2201) pwm_backlight: Init

W (2202) ledc: GPIO 0 is not usable, maybe conflict with others

E (2202) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position

I (2204) YellowDisplay: Display started successfully

E (2214) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position

W (2218) YellowDisplay: Failed to get backlight duty, using default value

E (2227) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position

I (2233) YellowDisplay: Backlight duty set to 50

E (2242) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position

I (2246) YellowDisplay: Backlight duty set to default 50

E (2255) lcd_panel: esp_lcd_panel_draw_bitmap(35): start position must be smaller than end position

I (2259) YellowDisplay: Backlight duty set to 0

assert failed: std::shared_ptr<tt::hal::display::DisplayDevice> tt::lvgl::initDisplay(const tt::hal::Configuration&) Init.cpp:40 (existing_display_user_data == nullptr)

Backtrace: 0x40081c55:0x3ffbedf0 0x4008a875:0x3ffbee10 0x4008f221:0x3ffbee30 0x400fe654:0x3ffbef50 0x400fe6be:0x3ffbef80 0x400e5051:0x3ffbefd0 0x400df392:0x3ffbf000 0x40231c82:0x3ffbf030 0x4008ac6a:0x3ffbf060

ELF file SHA256: 44e48d2c5
Rebooting...

```

NellowTCS, 2025-04-22:
So this version just added some logging so I can try to find out what is happening:
https://github.com/NellowTCS/Tactility/actions/runs/14605076699/artifacts/2989540903

NellowTCS, 2025-04-23:
I know that issue!
I'll fix it quickly and you can test it.

T-REX-XP, 2025-04-24:
> https://github.com/NellowTCS/Tactility/actions/runs/14624683301/artifacts/2996160926

Still blinking the skreen.

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6592
load:0x40078000,len:16656
load:0x40080400,len:4
load:0x40080404,len:4268
entry 0x40080658

I (29) boot: ESP-IDF v5.4 2nd stage bootloader
I (29) boot: compile time Apr 23 2025 17:49:47
I (29) boot: Multicore bootloader
I (30) boot: chip revision: v3.1
I (33) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00300000
I (81) boot:  3 system           Unknown data     01 81 00310000 00040000
I (88) boot:  4 data             Unknown data     01 81 00350000 00040000
I (95) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=6592ch (416044) map
I (225) esp_image: segment 1: paddr=00075954 vaddr=3ff80000 size=0001ch (    28) load
I (226) esp_image: segment 2: paddr=00075978 vaddr=3ffb0000 size=06028h ( 24616) load
I (237) esp_image: segment 3: paddr=0007b9a8 vaddr=40080000 size=04670h ( 18032) load
I (243) esp_image: segment 4: paddr=00080020 vaddr=400d0020 size=166694h (1468052) map
I (668) esp_image: segment 5: paddr=001e66bc vaddr=40084670 size=10e98h ( 69272) load
I (702) boot: Loaded app from partition at offset 0x10000
I (702) boot: Disabling RNG early entropy source...
I (713) cpu_start: Multicore app
I (721) cpu_start: Pro cpu start user code
I (721) cpu_start: cpu freq: 240000000 Hz
I (721) app_init: Application information:
I (721) app_init: Project name:     Tactility
I (725) app_init: App version:      0.4.0
I (729) app_init: Compile time:     Apr 23 2025 17:50:19
I (734) app_init: ELF file SHA256:  ecae0998a...
I (738) app_init: ESP-IDF:          v5.4
I (742) efuse_init: Min chip rev:     v0.0
I (745) efuse_init: Max chip rev:     v3.99 
I (749) efuse_init: Chip rev:         v3.1
I (753) heap_init: Initializing. RAM available for dynamic allocation:
I (760) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (765) heap_init: At 3FFBC668 len 00023998 (142 KiB): DRAM
I (770) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (775) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (781) heap_init: At 40095508 len 0000AAF8 (42 KiB): IRAM
I (787) spi_flash: detected chip: generic
I (790) spi_flash: flash io: qio
W (793) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (804) SDCard: Heap free before SD card init: 146872 bytes
I (808) SDCard: Heap free after SD card init: 146624 bytes
I (815) main_task: Started on CPU0
I (817) main_task: Calling app_main()
I (820) tactility: Tactility v0.4.0 on CYD 2432S022C (cyd-2432s022c)
I (833) tactility: nvs initialized
I (836) partitions: Mounted /system
[... 52 lines cut ...]
I (1836) service_registry: Adding sdcard
I (1840) service_registry: Starting sdcard
I (1844) service_registry: Started sdcard
I (1847) service_registry: Adding Wifi
I (1851) service_registry: Starting Wifi
I (1855) service_registry: Started Wifi
I (1858) service_registry: Adding EspNow
I (1862) service_registry: Starting EspNow
I (1866) service_registry: Started EspNow
I (1869) lvgl_init: Starting
I (1872) system_event: BootInitLvglBegin
I (1876) YellowTouch: Creating YellowTouch
I (1876) LVGL: Starting LVGL task
I (1880) YellowTouch: YellowTouch created: 0x3ffd2434
I (1887) YellowDisplay: DMA heap free: 173672
I (1891) YellowDisplay: Heap free before LVGL init: 173672

assert failed: xQueueGenericSend queue.c:936 (pxQueue)

Backtrace: 0x40081c55:0x3ffd56b0 0x4008a875:0x3ffd56d0 0x4008f221:0x3ffd56f0 0x402328be:0x3ffd5810 0x4010f597:0x3ffd5850 0x4008ac6a:0x3ffd5870

ELF file SHA256: ecae0998a

Rebooting...
```

NellowTCS, 2025-04-24:
> assert failed: xQueueGenericSend queue.c:936 (pxQueue)
> 
> Backtrace: 0x40081c55:0x3ffd56b0 0x4008a875:0x3ffd56d0 0x4008f221:0x3ffd56f0 0x402328be:0x3ffd5810 0x4010f597:0x3ffd5850 0x4008ac6a:0x3ffd5870
> 
> ELF file SHA256: ecae0998a
> 
> Rebooting...
> ```

Fortunately this is just a Mutex error. and it should be fixed now:
https://github.com/NellowTCS/Tactility/actions/runs/14645287012/artifacts/3003123116

T-REX-XP, 2025-04-25:
> Fortunately this is just a Mutex error. and it should be fixed now: https://github.com/NellowTCS/Tactility/actions/runs/14645287012/artifacts/3003123116

Still the same.
```
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6592
load:0x40078000,len:16656
load:0x40080400,len:4
load:0x40080404,len:4268
entry 0x40080658

I (29) boot: ESP-IDF v5.4 2nd stage bootloader
I (29) boot: compile time Apr 24 2025 15:19:42
I (29) boot: Multicore bootloader
I (30) boot: chip revision: v3.1
I (33) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00300000
I (81) boot:  3 system           Unknown data     01 81 00310000 00040000
I (88) boot:  4 data             Unknown data     01 81 00350000 00040000
I (95) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=6584ch (415820) map
I (225) esp_image: segment 1: paddr=00075874 vaddr=3ff80000 size=0001ch (28) load
I (225) esp_image: segment 2: paddr=00075898 vaddr=3ffb0000 size=06028h (24616) load
I (237) esp_image: segment 3: paddr=0007b8c8 vaddr=40080000 size=04750h (18256) load
I (243) esp_image: segment 4: paddr=00080020 vaddr=400d0020 size=166530h (1467696) map
I (668) esp_image: segment 5: paddr=001e6558 vaddr=40084750 size=10db8h (69048) load
I (702) boot: Loaded app from partition at offset 0x10000
I (702) boot: Disabling RNG early entropy source...
I (712) cpu_start: Multicore app
I (720) cpu_start: Pro cpu start user code
I (720) cpu_start: cpu freq: 240000000 Hz
I (721) app_init: Application information:
I (721) app_init: Project name:     Tactility
I (725) app_init: App version:      0.4.0
I (728) app_init: Compile time:     Apr 24 2025 15:20:15
I (733) app_init: ELF file SHA256:  a89d5b95f...
I (738) app_init: ESP-IDF:          v5.4
I (741) efuse_init: Min chip rev:   v0.0
I (745) efuse_init: Max chip rev:   v3.99 
I (749) efuse_init: Chip rev:       v3.1
I (753) heap_init: Initializing. RAM available for dynamic allocation:
I (759) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (764) heap_init: At 3FFBC660 len 000239A0 (142 KiB): DRAM
I (770) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (775) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (781) heap_init: At 40095508 len 0000AAF8 (42 KiB): IRAM
I (787) spi_flash: detected chip: generic
I (790) spi_flash: flash io: qio
W (793) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (804) SDCard: Heap free before SD card init: 146880 bytes
I (808) SDCard: Heap free after SD card init: 146632 bytes
I (814) main_task: Started on CPU0
I (816) main_task: Calling app_main()
I (819) tactility: Tactility v0.4.0 on CYD 2432S022C (cyd-2432s022c)
I (832) tactility: nvs initialized
I (834) partitions: Mounted /system
I (836) partitions: Mounted /data
I (838) system_event: BootInitHalBegin
I (838) system_event: BootInitI2cBegin
I (841) i2c: Init
I (843) i2c: (0) Started
I (845) system_event: BootInitI2cEnd
I (848) system_event: BootInitSpiBegin
I (852) spi: Init
I (854) spi: (2) Started
I (855) system_event: BootInitSpiEnd
I (859) system_event: BootInitUartBegin
I (862) uart: Init
I (864) system_event: BootInitUartEnd
I (867) hal: Init power
I (870) YellowTouch: Creating YellowTouch
I (874) YellowTouch: YellowTouch created: 0x3ffd0e64
W (878) YellowDisplay: [LOG] Entered YellowDisplay::start() at ./Boards/CYD-2432S022C/Source/hal/YellowDisplay.cpp:26

Backtrace: 0x400E2B2C:0x3FFBEE70 0x400E2155:0x3FFBEF00 0x400F9DA3:0x3FFBEF30 0x400E5024:0x3FFBEF90
           0x400E4F4C:0x3FFBEFC0 0x400DF37A:0x3FFBEFF0 0x40231B76:0x3FFBF020 0x4008AC6A:0x3FFBF050

I (905) YellowDisplay: DMA heap free: 189336
I (909) YellowDisplay: Heap free before LVGL init: 189336
I (914) YellowDisplay: Heap free after LVGL init: 189336
I (919) YellowDisplay: DMA heap free after LVGL init: 189336
I (925) YellowDisplay: Configured I80Display for 240x320
I (930) i80display: Starting I80 Display
I (934) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

❌ **ASSERT FAILED**  
`lvgl_port_lock esp_lvgl_port.c:159 (lvgl_port_ctx.lvgl_mux && "lvgl_port_init must be called first")`

Backtrace:  
0x40081c55:0x3ffbec50 0x4008a875:0x3ffbec70 0x4008f221:0x3ffbec90  
0x4010f2bd:0x3ffbedb0 0x4010fc95:0x3ffbedd0 0x400e49a9:0x3ffbee00  
0x400e4ae0:0x3ffbee50 0x400e2c94:0x3ffbee70 0x400e2155:0x3ffbef00  
0x400f9da3:0x3ffbef30 0x400e5024:0x3ffbef90 0x400e4f4c:0x3ffbefc0  
0x400df37a:0x3ffbeff0 0x40231b76:0x3ffbf020 0x4008ac6a:0x3ffbf050

ELF file SHA256: a89d5b95f
Rebooting...

```

NellowTCS, 2025-04-25:
So I decided to simplify it a bit and here is the hopefully fixed result:
https://github.com/NellowTCS/Tactility/actions/runs/14670663497/artifacts/3011598589

T-REX-XP, 2025-04-27:
> So I decided to simplify it a bit and here is the hopefully fixed result: https://github.com/NellowTCS/Tactility/actions/runs/14670663497/artifacts/3011598589

The screen just blinked once and then turned off.

```
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6592
load:0x40078000,len:16656
load:0x40080400,len:4
load:0x40080404,len:4268
entry 0x40080658
I (29) boot: ESP-IDF v5.4 2nd stage bootloader
I (29) boot: compile time Apr 25 2025 18:03:54
I (29) boot: Multicore bootloader
I (30) boot: chip revision: v3.1
I (33) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00300000
I (81) boot:  3 system           Unknown data     01 81 00310000 00040000
I (88) boot:  4 data             Unknown data     01 81 00350000 00040000
I (95) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=6555ch (415068) map
I (225) esp_image: segment 1: paddr=00075584 vaddr=3ff80000 size=0001ch (    28) load
I (225) esp_image: segment 2: paddr=000755a8 vaddr=3ffb0000 size=06028h ( 24616) load
I (237) esp_image: segment 3: paddr=0007b5d8 vaddr=40080000 size=04a40h ( 19008) load
I (243) esp_image: segment 4: paddr=00080020 vaddr=400d0020 size=166030h (1466416) map
I (667) esp_image: segment 5: paddr=001e6058 vaddr=40084a40 size=10ac8h ( 68296) load
I (701) boot: Loaded app from partition at offset 0x10000
I (701) boot: Disabling RNG early entropy source...
I (711) cpu_start: Multicore app
I (720) cpu_start: Pro cpu start user code
I (720) cpu_start: cpu freq: 240000000 Hz
I (720) app_init: Application information:
I (720) app_init: Project name:     Tactility
I (724) app_init: App version:      0.4.0
I (728) app_init: Compile time:     Apr 25 2025 18:04:25
I (733) app_init: ELF file SHA256:  92af153a1...
I (737) app_init: ESP-IDF:          v5.4
I (741) efuse_init: Min chip rev:     v0.0
I (744) efuse_init: Max chip rev:     v3.99 
I (748) efuse_init: Chip rev:         v3.1
I (752) heap_init: Initializing. RAM available for dynamic allocation:
I (759) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (764) heap_init: At 3FFBC660 len 000239A0 (142 KiB): DRAM
I (769) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (774) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (780) heap_init: At 40095508 len 0000AAF8 (42 KiB): IRAM
I (786) spi_flash: detected chip: generic
I (789) spi_flash: flash io: qio
W (792) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (803) SDCard: Heap free before SD card init: 146880 bytes
I (807) SDCard: Heap free after SD card init: 146632 bytes
I (813) main_task: Started on CPU0
I (815) main_task: Calling app_main()
I (818) tactility: Tactility v0.4.0 on CYD 2432S022C (cyd-2432s022c)
I (831) tactility: nvs initialized
I (833) partitions: Mounted /system
I (835) partitions: Mounted /data
I (837) system_event: BootInitHalBegin
[... 29 lines cut ...]
I (960) service_registry: Started sdcard
I (963) service_registry: Adding Wifi
I (966) service_registry: Starting Wifi
I (971) service_registry: Started Wifi
I (973) service_registry: Adding EspNow
I (977) service_registry: Starting EspNow
I (981) service_registry: Started EspNow
I (984) lvgl_init: Starting
I (987) system_event: BootInitLvglBegin
I (991) YellowDisplay: [LOG] Entered createDisplay() at ./Boards/CYD-2432S022C/Source/hal/YellowDisplay.cpp:18
I (991) LVGL: Starting LVGL task
I (1000) YellowTouch: Creating YellowTouch
I (1007) YellowTouch: YellowTouch created: 0x3ffd4af0
I (1012) YellowDisplay: Display config:
I (1015) YellowDisplay:   dataPins: [15 13 12 14 27 25 33 32]
I (1021) YellowDisplay:   dcPin: 16, wrPin: 4, csPin: 17, rstPin: -1, backlightPin: 0
I (1028) YellowDisplay:   Resolution: 240x320, Pixel Clock: 12000000 Hz, Bus Width: 8
I (1036) pwm_backlight: Init
I (1039) i80display: Starting I80 Display
I (1318) i80display: I80 Display initialization completed successfully
assert failed: std::shared_ptr<tt::hal::display::DisplayDevice> tt::lvgl::initDisplay(const tt::hal::Configuration&) Init.cpp:40 (existing_display_user_data == nullptr)
Backtrace: 0x40081c55:0x3ffbede0 0x4008a875:0x3ffbee00 0x4008f221:0x3ffbee20 0x400fe090:0x3ffbef40 0x400fe0fa:0x3ffbef70 0x400e4a8d:0x3ffbefc0 0x400df326:0x3ffbeff0 0x40231676:0x3ffbf020 0x4008ac6a:0x3ffbf050
ELF file SHA256: 92af153a1
Rebooting...
```

NellowTCS, 2025-04-27:
We're so close!!!
I keep doing this  `lv_display_set_user_data(displayHandle, this);`, which shouldn't be used as Tactility itself sets the display data.
This definitely should fix it:
https://github.com/NellowTCS/Tactility/actions/runs/14694562702/artifacts/3017691223

T-REX-XP, 2025-04-28:
> We're so close!!! I keep doing this `lv_display_set_user_data(displayHandle, this);`, which shouldn't be used as Tactility itself sets the display data. This definitely should fix it: https://github.com/NellowTCS/Tactility/actions/runs/14694562702/artifacts/3017691223

It's fixed now. There is no image on the screen. It's turned of, but seems like sensor works. I can see the new logs when touching the scren.

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:6592

load:0x40078000,len:16656

load:0x40080400,len:4

load:0x40080404,len:4268

entry 0x40080658

I (29) boot: ESP-IDF v5.4 2nd stage bootloader

I (29) boot: compile time Apr 27 2025 17:41:52

I (29) boot: Multicore bootloader

I (31) boot: chip revision: v3.1

I (33) qio_mode: Enabling default flash chip QIO

I (38) boot.esp32: SPI Speed      : 40MHz

I (41) boot.esp32: SPI Mode       : QIO

I (45) boot.esp32: SPI Flash Size : 4MB

I (49) boot: Enabling RNG early entropy source...

I (53) boot: Partition Table:

I (56) boot: ## Label            Usage          Type ST Offset   Length

I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (75) boot:  2 factory          factory app      00 00 00010000 00300000

I (82) boot:  3 system           Unknown data     01 81 00310000 00040000

I (88) boot:  4 data             Unknown data     01 81 00350000 00040000

I (95) boot: End of partition table

I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=6555ch (415068) map

I (225) esp_image: segment 1: paddr=00075584 vaddr=3ff80000 size=0001ch (    28) load

I (225) esp_image: segment 2: paddr=000755a8 vaddr=3ffb0000 size=06028h ( 24616) load

I (237) esp_image: segment 3: paddr=0007b5d8 vaddr=40080000 size=04a40h ( 19008) load

I (243) esp_image: segment 4: paddr=00080020 vaddr=400d0020 size=16602ch (1466412) map

I (667) esp_image: segment 5: paddr=001e6054 vaddr=40084a40 size=10ac8h ( 68296) load
[... 1036 lines cut ...]

I (7429) i80display: LVGL flush: area=0x3ffd3f10 x1=2 y1=266 x2=237 y2=275

I (7430) i80display: Batch draw: area=0x3ffd3f10 x1=2 y1=266 x2=237 y2=275

I (7442) i80display: LVGL flush: area=0x3ffd3f10 x1=2 y1=276 x2=237 y2=285

I (7443) i80display: Batch draw: area=0x3ffd3f10 x1=2 y1=276 x2=237 y2=285

I (7453) i80display: LVGL flush: area=0x3ffd3f10 x1=2 y1=286 x2=237 y2=295

I (7456) i80display: Batch draw: area=0x3ffd3f10 x1=2 y1=286 x2=237 y2=295

I (7479) i80display: LVGL flush: area=0x3ffd3f10 x1=2 y1=296 x2=237 y2=305

I (7480) i80display: Batch draw: area=0x3ffd3f10 x1=2 y1=296 x2=237 y2=305

I (7516) i80display: LVGL flush: area=0x3ffd3f10 x1=2 y1=306 x2=237 y2=315

I (7517) i80display: Batch draw: area=0x3ffd3f10 x1=2 y1=306 x2=237 y2=315

I (7522) i80display: LVGL flush: area=0x3ffd3f10 x1=2 y1=316 x2=237 y2=317

I (7525) i80display: Batch draw: area=0x3ffd3f10 x1=2 y1=316 x2=237 y2=317
```

NellowTCS, 2025-04-28:
🫠
I've had this same issue so many times, so I asked @KenVanHoeylandt  about it.
Still waiting on a response but try this as this explicitly sets the backlight after init:
https://github.com/NellowTCS/Tactility/actions/runs/14713610310/artifacts/3023804257

I'm really sorry about how long this is taking though!

NellowTCS, 2025-04-28:
After talking to him, I learned that may have the same issue so try this if it doesn't work:
https://github.com/NellowTCS/Tactility/actions/runs/14716054169/artifacts/3024653864

T-REX-XP, 2025-04-29:
> After talking to him, I learned that may have the same issue so try this if it doesn't work: https://github.com/NellowTCS/Tactility/actions/runs/14716054169/artifacts/3024653864

It's still a white screen. 
Here is the link to the logs - https://pastebin.com/jWtSrRVm

NellowTCS, 2025-05-03:
So this is a driver issue but I tried something:
https://github.com/NellowTCS/Tactility/actions/runs/14813852099/artifacts/3056065490

T-REX-XP, 2025-05-05:
> So this is a driver issue but I tried something: https://github.com/NellowTCS/Tactility/actions/runs/14813852099/artifacts/3056065490

It starts crashing right now.

**Logs:** [https://pastebin.com/aBbQu48b](https://pastebin.com/aBbQu48b)

```
I (999) LVGL: Starting LVGL task
I (1007) YellowTouch: Creating YellowTouch
I (1014) YellowTouch: YellowTouch created: 0x3ffd4b8c
I (1019) YellowDisplay: Display config:
I (1022) YellowDisplay:   dataPins: [15 13 12 14 27 25 33 32]
I (1028) YellowDisplay:   dcPin: 16, wrPin: 4, csPin: 17, rstPin: -1, backlightPin: 0
I (1035) YellowDisplay:   Resolution: 240x320, Pixel Clock: 12000000 Hz, Bus Width: 8
I (1043) i80display: Starting I80 Display
I (1047) i80display: Initializing I80 bus
I (1051) i80display: Configuring panel
 
assert failed: panel_io_i80_tx_param esp_lcd_panel_io_i2s.c:514 (param_size <= LCD_I80_IO_FORMAT_BUF_SIZE && "format buffer too small, increase LCD_I80_IO_FORMAT_BUF_SIZE")
```

NellowTCS, 2025-05-05:
Well that's kind of annoying but my fault sorry:
This new version should fix that and use the correct display init commands:
https://github.com/NellowTCS/Tactility/actions/runs/14847464194/artifacts/3065896101

T-REX-XP, 2025-05-06:
> Well that's kind of annoying but my fault sorry: This new version should fix that and use the correct display init commands: https://github.com/NellowTCS/Tactility/actions/runs/14847464194/artifacts/3065896101

The screen is blinking.
https://pastebin.com/9CgCj47A

```
I (1007) YellowTouch: Creating YellowTouch
I (1014) YellowTouch: YellowTouch created: 0x3ffd4b8c
I (1019) YellowDisplay: Display config:
I (1022) YellowDisplay:   dataPins: [15 13 12 14 27 25 33 32]
I (1028) YellowDisplay:   dcPin: 16, wrPin: 4, csPin: 17, rstPin: -1, backlightPin: 0
I (1035) YellowDisplay:   Resolution: 240x320, Pixel Clock: 12000000 Hz, Bus Width: 8
I (1043) i80display: Starting I80 Display
I (1047) i80display: Initializing I80 bus
I (1051) i80display: Configuring panel
I (1344) i80display: Setting up LVGL display
assert failed: lvgl_port_add_disp_priv esp_lvgl_port_disp.c:242 (disp_cfg != NULL)
 
Backtrace: 0x40081c55:0x3ffbed30 0x4008a875:0x3ffbed50 0x4008f221:0x3ffbed70 0x4010ec79:0x3ffbee90 0x4010f414:0x3ffbeed0 0x400e3de5:0x3ffbef00 0x400e4274:0x3ffbef20 0x400fdc7a:0x3ffbef40 0x400fdd52:0x3ffbef70 0x400e46e5:0x3ffbefc0 0x400df312:0x3ffbeff0 0x402313ca:0x3ffbf020 0x4008ac6a:0x3ffbf050
 
ELF file SHA256: 8d73b3c89
```

NellowTCS, 2025-05-07:
This should fix that:
https://github.com/NellowTCS/Tactility/actions/runs/14873410616/artifacts/3074886909

T-REX-XP, 2025-05-13:
Still blinking screen, I mean crashing

```
 I (1014) YellowTouch: YellowTouch created: 0x3ffd4c94

I (1019) YellowDisplay: Display config:

I (1022) YellowDisplay:   dataPins: [15 13 12 14 27 25 33 32]

I (1028) YellowDisplay:   dcPin: 16, wrPin: 4, csPin: 17, rstPin: -1, backlightPin: 0

I (1035) YellowDisplay:   Resolution: 240x320, Pixel Clock: 12000000 Hz, Bus Width: 8

I (1043) i80display: Starting I80 Display

I (1047) i80display: Initializing I80 bus

I (1051) i80display: Configuring panel

I (1344) i80display: Setting up LVGL display

E (1344) LVGL: lvgl_port_add_disp_priv(312): Not enough memory for LVGL buffer (buf1) allocation!

E (1345) i80display: Failed to create LVGL display

E (1350) i80display: Failed to set up LVGL display

I (1354) i80display: I80 Display stopped successfully

E (1359) lvgl_init: Display start failed

I (1363) i80display: Stopping I80 Display

W (1366) i80display: Display not started

I (1370) tactility: Registering and starting system services

I (1376) service_registry: Adding Loader

I (1379) service_registry: Starting Loader

I (1384) Thread: Starting loader_dispatcher

I (1387) service_registry: Started Loader

I (1391) service_registry: Adding Gui

I (1394) service_registry: Starting Gui

assert failed: tt::service::gui::Gui* tt::service::gui::gui_alloc() Gui.cpp:42 (screen_root != nullptr)

Backtrace: 0x40081c55:0x3ffbee80 0x4008a875:0x3ffbeea0 0x4008f221:0x3ffbeec0 0x401021ae:0x3ffbefe0 0x40102241:0x3ffbf040 0x400ff10b:0x3ffbf060 0x400ff1b4:0x3ffbf0a0 0x400ff1d4:0x3ffbf0e0 0x400e47a5:0x3ffbf110 0x400e4908:0x3ffbf130 0x400df33e:0x3ffbf160 0x402316da:0x3ffbf190 0x4008ac6a:0x3ffbf1c0
ELF file SHA256: fc1e7794e

Rebooting...
```
Full log: https://pastebin.com/wWyA17nc

NellowTCS, 2025-05-14:
> E (1344) LVGL: lvgl_port_add_disp_priv(312): Not enough memory for LVGL buffer (buf1) allocation!

Having some deja vu but this should fix that:
https://github.com/NellowTCS/Tactility/actions/runs/15028558675/artifacts/3125458699

T-REX-XP, 2025-05-22:
> Having some deja vu but this should fix that:
> https://github.com/NellowTCS/Tactility/actions/runs/15028558675/artifacts/3125458699

Still crashing

```
I (994) system_event: BootInitLvglBegin
I (998) YellowDisplay: [LOG] Entered createDisplay() at ./Boards/CYD-2432S022C/Source/hal/YellowDisplay.cpp:18
I (998) LVGL: Starting LVGL task
I (1007) YellowTouch: Creating YellowTouch
I (1014) YellowTouch: YellowTouch created: 0x3ffd4c94
I (1019) YellowDisplay: Display config:
I (1022) YellowDisplay:   dataPins: [15 13 12 14 27 25 33 32]
I (1028) YellowDisplay:   dcPin: 16, wrPin: 4, csPin: 17, rstPin: -1, backlightPin: 0
I (1035) YellowDisplay:   Resolution: 240x320, Pixel Clock: 12000000 Hz, Bus Width: 8
I (1043) i80display: Starting I80 Display
I (1047) i80display: Initializing I80 bus
I (1051) i80display: Configuring panel
I (1344) i80display: Setting up LVGL display
E (1344) LVGL: lvgl_port_add_disp_priv(312): Not enough memory for LVGL buffer (buf1) allocation!
E (1345) i80display: Failed to create LVGL display
E (1350) i80display: Failed to set up LVGL display
I (1354) i80display: I80 Display stopped successfully
E (1359) lvgl_init: Display start failed
I (1363) i80display: Stopping I80 Display
W (1366) i80display: Display not started
I (1370) tactility: Registering and starting system services
I (1376) service_registry: Adding Loader
I (1379) service_registry: Starting Loader
I (1384) Thread: Starting loader_dispatcher
I (1387) service_registry: Started Loader
I (1391) service_registry: Adding Gui
I (1394) service_registry: Starting Gui
assert failed: tt::service::gui::Gui* tt::service::gui::gui_alloc() Gui.cpp:42 (screen_root != nullptr)
Backtrace: 0x40081c55:0x3ffbee80 0x4008a875:0x3ffbeea0 0x4008f221:0x3ffbeec0 0x401021b2:0x3ffbefe0 0x40102245:0x3ffbf040 0x400ff10f:0x3ffbf060 0x400ff1b8:0x3ffbf0a0 0x400ff1d8:0x3ffbf0e0 0x400e47a9:0x3ffbf110 0x400e490c:0x3ffbf130 0x400df33e:0x3ffbf160 0x402316de:0x3ffbf190 0x4008ac6a:0x3ffbf1c0
ELF file SHA256: b2b6a7bf1
Rebooting...
```

Here is the full logs: https://pastebin.com/KVPwRz97

NellowTCS, 2025-05-22:
Weird, the buffer is already tiny:
Made even tinier:
https://github.com/NellowTCS/Tactility/actions/runs/15193966638/artifacts/3179586534

NellowTCS, 2025-07-21:
As this has gotten really long, I decided to move it to the fork ^