TactilityProject/Tactility #274: CYD E32R28T One USB-C

KlutzyBlade, 2025-04-08:
I have the board below and Id love to run Tactility on it. Its similar to the CYD2usb and I can run Launcher, Ghost, and Bruce on it. I like Tactility being able to tell me what GPIO's are available to be able to play with different modules. I would greatly appreciate your help.

Back Photo - http://www.lcdwiki.com/2.8inch_ESP32-32E_Display#/media/File:E32R28T-04.png
ESP32 - http://www.lcdwiki.com/2.8inch_ESP32-32E_Display#ESP32_Parameters
Overall Info Dump Link - http://www.lcdwiki.com/2.8inch_ESP32-32E_Display

KenVanHoeylandt, 2025-04-08:
Hi! Someone would need to implement the drivers to support the board. I don't own that board, so I'm afraid I can't implement it for you.

That said, the display is ILI9341V, which has a compatible [driver](https://github.com/ByteWelder/Tactility/tree/main/Drivers/ILI934x) already implemented in Tactility. Similarly, the touch driver is XPT2046, which is also [available](https://github.com/ByteWelder/Tactility/tree/main/Drivers/XPT2046) within the project.

NellowTCS, 2025-04-09:
> http://www.lcdwiki.com/2.8inch_ESP32-32E_Display

Lookin at the documentation, that board seems to be super similar to the ESP32-2432S028R, so it should work with that build afaik.
I am working on implementing that and have a fork, but the touchscreen is currently my biggest hurdle (getting it calibrated).
You can try out that build with the attached firmware or get it from [here.](https://github.com/NellowTCS/Tactility/actions/runs/14344679438)
[Tactility-cyd-2432s028r.zip](https://github.com/user-attachments/files/19659158/Tactility-cyd-2432s028r.zip)

Let me know if it's not calibrated!

KlutzyBlade, 2025-04-09:
Thank you both for your help! @NellowTCS I have downloaded the firmware and tried to flash it to my board with the espressif Flash tool and received the error below. Could you point me to a resource that could help me understand how to setup platformio to upload things to my board? Also, I would love to ask either of you a few, probably noob, questions about this process that I have been unable to find answers for online. I'd be glad to chat with you if you have the time. Thanks again!

NellowTCS, 2025-04-09:
> Thank you both for your help! [@NellowTCS](https://github.com/NellowTCS) I have downloaded the firmware and tried to flash it to my board with the espressif Flash tool and received the error below. [...]

No problem!
So you are currently flashing Tactility.bin, which is all of the data and images (afaik).
You are supposed to run ./flash.sh from the command line (something like `./flash.sh  [the port of your esp32]`. Also, (you might already be doing this), but you are supposed to have esptool and use that (also from the command line).
For a much better guide you can check out [the Tactility website](https://tactility.one/#/installing?id=releases-manual-flashing).
Also, sure! If you need any help, just reach out!

KlutzyBlade, 2025-04-09:
Hi! I was able to use the bin files in your download and it uploaded to my board but the touch does not work. It also doesn't look like its taking up the full screen. Hopefully the photo below will help a little.

NellowTCS, 2025-04-09:
> Hi! I was able to use the bin files in your download and it uploaded to my board but the touch does not work. It also doesn't look like its taking up the full screen. Hopefully the photo below will help a little.

That's pretty useful!
So, the full display is being used, but the device doesn't have enough RAM to display icons, and so the display looks partially unused.

Try removing the microSD card if you have one plugged in.

Could you share the serial logs from a fresh reboot, and try touching the display during that?

KlutzyBlade, 2025-04-09:
Id love too but its gibberish in Arduino Serial Monitor. How do you usually get that info?

NellowTCS, 2025-04-09:
> Id love to but its gibberish in Arduino Serial Monitor. How do you usually get that info?

I used https://serial.huhn.me as it seems to work the best with Tactility.

NellowTCS, 2025-04-10:
> [image]

Did you touch the display during the logs, as the constant `Touch raw (rotated):` is due to an IRQ bug and not (that) related to touch?
Also the calibration on the below build should be closer, but you may still have to move around a bit to get it to click stuff.

[Tactility-cyd-2432s028r.zip](https://github.com/user-attachments/files/19685849/Tactility-cyd-2432s028r.zip)

KlutzyBlade, 2025-04-10:
Awesome thanks!

NellowTCS, 2025-04-10:
> I loaded the latest build and it works, the touch is still off as you said. Is there any way I can help you calibrate your fork with my CYD?

I am not sure to be honest, as I made a mistake with the Calibration app I made (first dot goes off the screen) and I need to confirm something before fixing it.

You could try touching each corner of the display, and sending those logs, so that I can confirm with two sets of logs.

Also, are app icons currently visible on your display?

Thanks for the help!

KlutzyBlade, 2025-04-10:
> I am not sure to be honest, as I made a mistake with the Calibration app I made (first dot goes off the screen) and I need to confirm something before fixing it.
>
> You could try touching each corner [...]

There are no icons on my display only text options. Here's my touch coordinates.

[CYD_Single_USBC_Touch_Output.pdf](https://github.com/user-attachments/files/19686932/CYD_Single_USBC_Touch_Output.pdf)

NellowTCS, 2025-04-10:
> There are no icons on my display only text options. Here's my touch coordinates.
> 
> [CYD_Single_USBC_Touch_Output.pdf](https://github.com/user-attachments/files/19686932/CYD_Single_USBC_Touch_Output.pdf)

Alright that means your display doesn't have enough RAM still.
I'll fix that later, but the coordinates are super useful!

NellowTCS, 2025-04-10:
More calibration 👍:

[Tactility-cyd-2432s028r.zip](https://github.com/user-attachments/files/19689320/Tactility-cyd-2432s028r.zip)

KlutzyBlade, 2025-04-10:
> More calibration 👍:
> 
> [Tactility-cyd-2432s028r.zip](https://github.com/user-attachments/files/19689320/Tactility-cyd-2432s028r.zip)

Heres the touch info for calibration. I did top left, bottom left, top right, bottom right. The screen was in portrait orientation with the charging port down.

[CYD_SingleUSB-C_Touch_Output_mkII.docx](https://github.com/user-attachments/files/19693272/CYD_SingleUSB-C_Touch_Output_mkII.docx)

NellowTCS, 2025-04-12:
here's some more calibration:

[Tactility-cyd-2432s028r.zip](https://github.com/user-attachments/files/19717479/Tactility-cyd-2432s028r.zip)

NellowTCS, 2025-04-14:
This should be much better (I haven't been able to test this version yet) as I am trying auto calibration.

[Tactility-cyd-2432s028r.zip](https://github.com/user-attachments/files/19728757/Tactility-cyd-2432s028r.zip)

NellowTCS, 2025-04-14:
> Thanks so much for the help! The download with more calibration had around the same results as the others unfortunately. The latest download only registered one touch in serial terminal and would not register any further touch inputs.

The newest version also gave me the same issue, I'll try to fix it. 👍

NellowTCS, 2025-04-14:
This should be better:

[Tactility-cyd-2432s028r.zip](https://github.com/user-attachments/files/19743896/Tactility-cyd-2432s028r.zip)

NellowTCS, 2025-04-19:
[Tactility-cyd-2432s028r.zip](https://github.com/NellowTCS/Tactility/actions/runs/14544593699/artifacts/2972541349)

I have the device and I did a major rewrite of a lot of stuff, so this should work better.

NellowTCS, 2025-07-21:
Continued due to size at 
https://github.com/NellowTCS/Tactility/issues/4

closed 2025-07-22 by KenVanHoeylandt