TactilityProject/Tactility #294: Add basic support for LilyGO T-Lora Pager

fipfip, 2025-06-30:
Before creating a PR, I'd like to discuss the WIP which adds basic support for the [LilyGO T-Lora Pager](https://github.com/Xinyuan-LilyGO/LilyGoLib/blob/master/docs/hardware/lilygo-t-lora-pager.md).  
It is a tiny ESP32-S3 based pager-like, with limited input through a keyboard and encoder wheel.  

Currently hardware support for the following is implemented:
 - ST7796 Display
 - Backlit keyboard via TCA8418
 - Encoder wheel via ESP pulse counter peripheral
 - Power status via BQ27220
 - GNSS via MIA-M10Q
 - SD/TF Card via SPI

The following hardware is intentionally not supported yet, but might be subject to discussion:
 - LoRa: Semtech SX1262
 - NFC: ST25R3916
 - Smart sensor: Bosch BHI260AP
 - Real-Time Clock: NXP PCF85063A
 - Battery Charger: Ti BQ25896
 - Haptic driver: Ti DRV2605
 - Audio Codec: Everest-semi ES8311
 - GPIO Expand: XINLUDA XL9555

Some minor app changes are also required to make encoder input as convenient as possible, which will be subject for a future PR.
 
 # Tested on
 -  Hardware: LilyGO T-Lora Pager K257 (SX1262, 868MHz)
 - Tactility version: v0.4.0
 -  ESP-IDF version: ESP-IDF 5.4

# Changes
 - Board: files for board specific implementations
 - Tactilty:
   * service/gui/GuiDraw: Added encoder wheel support by adding default group
   * hal/i2c/I2cDevice: Added more bus access methods to implement BQ27220
 - Drivers: Add implementation for BQ27220, TCA8418, ST7796
 - Libraries: Add esp_lcd_st7796 library

# Code
 - Local Mirror: https://github.com/fipfip/Tactility/tree/pager
 - Origin: https://git.hoeglinger.eu/dominic/Tactiliest/src/branch/pager/ 

# Photos

[image]
Launcher, note that it is modified with uncomitted changes.

[image]
Notes app with a pangram.

[image]
Power settings showing status of BQ27220.

KenVanHoeylandt, 2025-07-01:
Thank you for your work to get this working!

> The following hardware is intentionally not supported yet, but might be subject to discussion:

This is not a problem. I have plenty of devices currently that don't offer a full hardware implementation. The basics are good to start with!

> Some minor app changes are also required to make encoder input as convenient as possible, which will be subject for a future PR.

👍 

> Drivers: Add implementation for BQ27220, TCA8418, ST7796

That's awesome! Each driver can help us to implement more devices in the future.

Feel free to open a pull request. I'll be happy to review it asap.

fipfip, 2025-07-01:
>This is not a problem. I have plenty of devices currently that don't offer a full hardware implementation. The basics are good to start with!

Thank you, tough there are already ideas in my head on making a LoRa service for shared access of the SX1262 between apps. Do you prefer if I open an issue for discussing this? This issue is fine also of course.

>I took a look at [your implementation](https://github.com/fipfip/Tactility/commit/9e6c60bc145ff503263aeb66d69f81a740bae34c#diff-dc86154e02477918008c17444e8016c4c105ab34c0f0d59debd945b777301151) and only have 1 question:
>
>Was it required to copy in [Libraries/esp_lcd_st7796](https://github.com/fipfip/Tactility/commit/9e6c60bc145ff503263aeb66d69f81a740bae34c#diff-6a86ec1ed7f4ccb5823947a6b6cbe901dbc4bc5e513fe3c1e5049a53ddbedda4) (e.g. to do some changes), as opposed to adding it at [App/idf_component.yml](https://github.com/ByteWelder/Tactility/blob/main/App/idf_component.yml)?

My bad, I thought libraries are managed that way but I see now given I couldn't add the dependency via idf.py.
Thank you for showing me the way to do it, I have updated my commit and I think its ready for a pull request.