TactilityProject/Tactility #297: Custom board keyboard and sdcard issue

DRIFTYY777, 2025-07-13:
I created a custom board using esp32s3. Boards contains Display, Sdcard, radio(NRF24), RFID and more....
Board dosent have a full keyboard but have D pad like structure or button style like flipper zero up, down, left, right, ok, and back connected to IO Expander (PCA9555) I tried writing pca9555 drivers its worked when I tested them but failed when I used with this project.
Logs -
I (1492) devices: Registered ESP32S3-ZERO Keyboard with id 2
I (1492) lvgl_init: Keyboard init
I (1495) esp32s3_zero_keyboard: Checking for keyboard at I2C address 0x20 on bus 0
E (1602) esp32s3_zero_keyboard: ESP32S3-ZERO keyboard not detected: ESP_ERR_TIMEOUT
E (1602) lvgl_init: Keyboard attach failed
I (1602) lvgl_init: Finished

Still I dont know ho to debug logs but It failed and I can't figured it out why! becuase I'm using D pad?

And the other with my sdcard thats my fault too "I thought" becuase I used non hardware spi thus making this problem or I just formated it wrong way 
Logs -
I (1192) hal: Mounting sdcard
I (1194) gpio: GPIO[39]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1203) spi_sdcard: Mounting /sdcard
E (1240) vfs_fat_sdmmc: sdmmc_card_init failed (0x105).
W (1287) hal: SD card mount failed (init can continue)

Again its a bad decision using non harware spi becuase I want 4-5 devices on spi bus and I dont want to load one bus.

Here's the fork 
https://github.com/DRIFTYY777/Tactility-Zero/tree/main/Boards/ESP32S3-ZERO

[ESP32s3 Zero Config.txt](https://github.com/DRIFTYY777/Tactility-Zero/blob/main/Boards/ESP32S3-ZERO/ESP32s3%20Zero%20Config.txt) Contains board hardware details

Somebody help 😑

KenVanHoeylandt, 2025-07-14:
### I2C double init

I see you're caling `init_i2c_master()` yourself in `ESP32S3ZeroKeyboard.cpp`. Tactility does that for you with the parameters that you've put in `esp32s3_zero_config`'s `.i2c` configuration. So you can remove your calls to `init_i2c_master()`.

The double init is likely causing issues.

### I2C native vs Tactility calls

I noticed that you're using `i2c_master_start()` etc
This is not necessarily problematic by itself, but it could be in the future: it's not thread-safe with regards to other I2C devices on that same bus. Using `tt::i2c::masterWrite()` and similar calls is thread-safe.

e.g. if you have a keyboard attached and you open the Scanner app, then both might be using the I2C bus at the same time and cause problems.

### I2C device detection

To detect devices, you can use `tt::i2c::masterHasDeviceAtAddress()`

To test I2C, I made a scanner app. If you don't have a working touch screen or other input means, you can automatically launch it on startup: Open `App/Source/Main.cpp` and edit the `tt::Configuration` by adding `.autoStartAppId = "I2cScanner"`
If it doesn't show up here, then the I2C configuration is likely incorrect.

If the scanner app selects the wrong bus on startup, you can change `I2cScannerApp::selectBus()` at the bottom of the function `I2cScannerApp::onShow()`

### SPI config

`configuration->spiFrequencyKhz = 25000000`

This is the likely culprit: 25 GHz is too fast.

### SPI functions

If you don't use `tt::hal::spi::masterWrite()` but use the esp-idf native calls directly then make sure to use `tt::hal::spi::getLock(device)` and acquire the lock afterwards. This ensures that you don't have 2 operations in parallel using the bus at the same time.

You can use the native esp-idf read/write calls for now - with that Tactility lock - as I haven't wrapped them yet into `tt::hal::spi::` yet.

DRIFTYY777, 2025-07-14:
Thanks for your analysing` (25 GHz)` and recommendations
I tried inbuild methods for I2c but still the same result 
`E (1457) lvgl_init: Keyboard attach failed`
Reconfig the I2c init
change direct methods to` tt::i2c`
And alos I updated the PCA9555 drivers example which was working [main.c](https://github.com/DRIFTYY777/Tactility-Zero/blob/main/Drivers/PCA9555/example/main.c)

KenVanHoeylandt, 2025-07-15:
Anything visible in the I2C Scanner app?

DRIFTYY777, 2025-07-15:
Nope