TactilityProject/Tactility #309: CYD-2.2" - Need Tester

NellowTCS, 2025-08-30:
I am working on an implementation for the 2.2" CYD (ESP32-2432S022C)
But I have a (what seems to be) a clone of this display, and so am not sure if it will work with other ones.

Anyone here have this display?

If so, could you try this build out?
(In order to flash it check out the docs [here](https://tactility.one/#/installing?id=releases-manual-flashing))
https://github.com/NellowTCS/Tactility/actions/runs/17347946561/artifacts/3890273559

KenVanHoeylandt, 2025-09-06:
It crashes on boot because of an assert:

> I (1118) EspLcdTouch: Adding touch to LVGL
> 
> assert failed: lvgl_port_add_touch esp_lvgl_port_touch.c:40 (touch_cfg->disp != NULL)

KenVanHoeylandt, 2025-09-06:
If you have time, consider taking a look at how I updated your latest board implementation:
https://github.com/ByteWelder/Tactility/tree/main/Boards/CYD-E32R28T/Source

- Renamed `hal/` to `devices/`
- Renamed things like `YellowSdCard` to `SdCard` (files and methods)
- Updated `tt::hal::Configuration` with `.createDevices = ` instead of manually assigning display, sd card etc.
- Cleaned up the [createDisplay()](https://github.com/ByteWelder/Tactility/blob/main/Boards/CYD-E32R28T/Source/devices/Display.cpp) implementation (prober allocation, no manual call to `start()`)

KenVanHoeylandt, 2025-09-06:
Keep in mind that there's a possibility that I have a different board that is named the same.

NellowTCS, 2025-09-06:
So it must have been a really lucky (or unlucky) startup as I was doing this 🤦‍♂️ :
```
.max_transfer_bytes = static_cast<size_t>(configuration->horizontalResolution * configuration->verticalResolution * 2),
```
(The entire screen being buffered)

Fixed that:
https://github.com/NellowTCS/Tactility/actions/runs/17516690493/artifacts/3944307042

NellowTCS, 2025-09-06:
Added those:
https://github.com/NellowTCS/Tactility/actions/runs/17519814705/artifacts/3945058110

KenVanHoeylandt, 2025-09-07:
It doesn't crash anymore. The display now shows a white screen permanently. Touch does work and you can launch apps, which I verified by looking at the logging in the serial monitor. SD card also mounts.

NellowTCS, 2025-09-07:
Added a massive debug log:
https://github.com/NellowTCS/Tactility/actions/runs/17531311559/artifacts/3947898881

NellowTCS, 2025-09-10:
> > W (820) spi_flash: boya detected but related driver is not linked, please check option `SPI_FLASH_SUPPORT_BOYA_CHIP`

That's just a chip-specific config not added, right? The flash is still detected (but as Generic) and used?
Either way, fixed:
https://github.com/NellowTCS/Tactility/actions/runs/17628584785/artifacts/3980554603

NellowTCS, 2025-09-13:
@KenVanHoeylandt  mind adding `help wanted` to this?
I am going to try to get that wrapper working with only the absolutely required stuff, but it'd be super helpful if someone else knows about this board better than me!

NellowTCS, 2025-09-14:
This was surprisingly easier than I thought:
https://github.com/NellowTCS/Tactility/actions/runs/17704775473/artifacts/4004698405

(Lovyan)

T-REX-XP, 2025-09-14:
> This was surprisingly easier than I thought: https://github.com/NellowTCS/Tactility/actions/runs/17704775473/artifacts/4004698405
> 
> (Lovyan)

Sensor doesn't work. Screen shows garbage.

[image]

```
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6704
load:0x40078000,len:16856
load:0x40080400,len:4288
entry 0x40080658
I (27) boot: ESP-IDF v5.5 2nd stage bootloader

I (27) boot: compile time Sep 14 2025 01:47:27

I (28) boot: Multicore bootloader

I (29) boot: chip revision: v3.1

I (31) qio_mode: Enabling default flash chip QIO

I (36) boot.esp32: SPI Speed      : 40MHz

I (39) boot.esp32: SPI Mode       : QIO

I (43) boot.esp32: SPI Flash Size : 4MB

I (47) boot: Enabling RNG early entropy source...

I (51) boot: Partition Table:

I (54) boot: ## Label            Usage          Type ST Offset   Length

I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (73) boot:  2 factory          factory app      00 00 00010000 00300000

I (80) boot:  3 system           Unknown data     01 81 00310000 0004b000

I (86) boot:  4 data             Unknown data     01 81 0035b000 00096000

I (93) boot: End of partition table

I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=68498h (427160) map

I (227) esp_image: segment 1: paddr=000784c0 vaddr=3ff80000 size=00020h (    32) load

I (227) esp_image: segment 2: paddr=000784e8 vaddr=3ffb0000 size=05ed0h ( 24272) load

I (238) esp_image: segment 3: paddr=0007e3c0 vaddr=40080000 size=01c58h (  7256) load

I (241) esp_image: segment 4: paddr=00080020 vaddr=400d0020 size=17c3a8h (1557416) map

I (695) esp_image: segment 5: paddr=001fc3d0 vaddr=40081c58 size=13aech ( 80620) load

I (733) boot: Loaded app from partition at offset 0x10000

I (733) boot: Disabling RNG early entropy source...

I (744) cpu_start: Multicore app

[... 376 lines cut ...]
I (2715) Loader: App "Boot" state: showing -> hiding

I (2715) Loader: App "Boot" state: hiding -> stopped

I (2721) PropertiesFile: Reading properties file /data/settings/boot.properties

I (2729) Loader: Start app Launcher

I (2730) Thread: Stopped boot

I (2739) Loader: Free heap: 218064

I (2739) Loader: Start by id Launcher

I (2740) Loader: App "Launcher" state: initial -> initial

I (2744) Loader: App "Launcher" state: initial -> started

I (2756) PropertiesFile: Reading properties file /data/settings/boot.properties

I (2758) Loader: App "Launcher" state: started -> showing

Input

```

KenVanHoeylandt, 2025-09-14:
> Followed the openHASP implementation better:
Same distorted visual.

NellowTCS, 2025-09-18:
So currently LovyanGFX is a submodule and not modified. It's ESP-IDF compatible (which I hadn't known before), and so it fits in perfectly. I have a bit of a wrapper around it to add some methods and wrap it with the Tactility specific things.

But I lowered the clock speed, _might_ fix it:
https://github.com/NellowTCS/Tactility/actions/runs/17816159167/artifacts/4041173233

KenVanHoeylandt, 2025-09-18:
Same issue.

NellowTCS, 2025-09-22:
https://github.com/NellowTCS/Tactility/actions/runs/17929519494/artifacts/4077054220

Reverted to a custom driver with the init commands from the LovyanGFX driver.

NellowTCS, 2025-09-24:
Reworked it a bit
https://github.com/NellowTCS/Tactility/actions/runs/17964801442/artifacts/4089408351

NellowTCS, 2026-10-10:
Closed as completed by #657

closed 2026-10-10 by NellowTCS