TactilityProject/Tactility #315: Bug: External App - Error when building Hello World

Revers-BR, 2025-09-02:
**Describe the bug**

When trying to build an external app, it returns an error in the Tactility SDK version regarding ESP-IDF

**To Reproduce**

Tactility/ExternalApps/HelloWorld$ python3 tactility.py build esp32

Tactility Build System v1.2.1
Downloading SDK version 0.4.0 for esp32
Building first build
Executing action: all (aliases: build)
................

CMake Error at .tactility/0.4.0-esp32/TactilitySDK/TactilitySDK.cmake:15 (message):
  ESP-IDF version of Tactility SDK (5.4) does not match current ESP-IDF
  version (5.5)
Call Stack (most recent call first):
  CMakeLists.txt:16 (tactility_project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/rivai/Tactility/ExternalApps/HelloWorld/build-esp32/log/idf_py_stderr_output_520288 and /home/rivai/Tactility/ExternalApps/HelloWorld/build-esp32/log/idf_py_stdout_output_520288
Building for esp32 failed ❌

**Details (please complete the following information):**
 - ESP type: ESP32
 - Device:  Ubuntu 20.04 (WSL)
 - Tactility version: main branch]

Revers-BR, 2025-09-03:
Calculator compiles without problem

Revers-BR, 2025-09-06:
> **Describe the bug**
>
> When trying to build an external app, it returns an error in the Tactility SDK version regarding ESP-IDF
>
> **To Reproduce**
>
> Tactility/ExternalApps/HelloWorld$ python3 [...]

Hello world app is compiling and working now

KenVanHoeylandt, 2025-09-06:
> /home/rivai/Tactility/ExternalApps/GraphicsDemo/main/Include/drivers/DisplayDriver.h:25:34: error: 'MAX_TICKS' 

That means it's not using the latest SDK.
Make sure to remove `release/`, then run `Buildscript/release-sdk-current.sh` from Tactility project root.

Then you can navigate to `ExternalApps/GraphicsDemo`, run `export TACTILITY_SDK_PATH=../../release` and you're set.

Revers-BR, 2025-09-06:
> Can you verify that you have the very latest code? I pushed some fixes an hour ago. (`Xpt2046SoftSPi` returned `true` for `supportsTouchDriver()` but `getTouchDriver()` returned `nullptr`)

Now it's working! @KenVanHoeylandt

closed 2025-09-07 by KenVanHoeylandt