TactilityProject/Tactility #342: Enhancement: Radio Support

fipfip, 2025-09-20:
Hi everone

In this issue I propose a new API for radio devices.
This would not only make Tactility more capable, but also open the door to supporting radio applications
such as a Meshcore or Meshtastic clients.

## Architecture

To facilitate different libraries, the configuration is done in two steps:
1. Load modem parameters which are enumerated into a device and
2. Start the modem in its desired mode which will apply the previously set parameters

Parameters can be both common like center frequency and power, but also specialized such as the SX1262's LR-FHSS's "Narrow Grid" option. All parameters are float, as this can be easily cast to any other scalar type whilst also avoiding overhead. Plus, keep it simple, right?

Since setting sets of parameters can be cumbersome, especially regarding presets, the helper class "ParameterSet" is included which bundles any combination of them into an unordered map.

Receiving packets is done via the PubSub pattern as established in the GPS device.
Transmitting packets is handled via a classic queue, with a callback function for status notification.

Currently development only included support for the SX1262 by Semtech, but since RadioLib is the backbone more devices can easily be implemented using the Sx1262 class as a reference.

## Tests

Currently the implementation is tested with a custom test app "ChirpChatter" (name stolen from SDRangel's ChirpChat) by transceiving LoRa.
This application will not be included in a future PR, it is more of a quickly hacked together demo application.
The hardware is a LilyGo T-Lora Pager, 868MHz model.

RX has been demonstrated a while ago privately via a mail to Ken and possibly shared on the Tactility Discord, see attached video "LoRa RX Demo".
Now, TX was also tested successfully using both raw and Meshtastic packets, see "Meshtastic TX Demo".

There are still reliability issues, if anything stands out any hints would be greatly appreciated.

⚠️ Disclosure and disclaimer: Meshtastic packets are pre-rendered for now in ChirpChatter.
It is highly advisable to keep traffic in regulation with local laws, as the application does not respect any rate limits.

## Code

Currently the code is available for a first look, and possibly pre-review here:
https://git.hoeglinger.eu/dominic/Tactiliest/src/branch/radio/
This Gitea instance support SSO via Github if anyone wants to participate here.

Open TODOs:
 - C API for use with external apps
 - Stabilization of SX1262 driver

Further development will resume once the architecture has been approved.

## Attachments
LoRa RX Demo: https://github.com/user-attachments/assets/c0c48ad8-53df-4941-8421-a5d905287da0
Meshtastic TX Demo: https://github.com/user-attachments/assets/b3fae01f-465e-4360-9159-a2c2d71d69dd

KenVanHoeylandt, 2025-09-20:
Thanks Dominic, that looks really good!

The videos are very low-res, so I couldn't make up a lot of details (GitHub probably compressed them too much)

I took a look at your branch in these areas:
- `Boards/LilygoTLoraPager/`
- `Drivers/RadioLibCompat/` and `Drivers/SX126x/`
- `tt::hal::radio`-related files
(and  `ChirpChatterApp.cpp`)
Let me know if I missed something relevant!

The architecture looks solid to me.
I'll have some minor remarks here and there, in an in-depth review.
e.g. `RadiolibTactilityHal::spiXXTransaction()` functions might need to use `hal::spi::getLock()`. The SPI bus is shared with the display, and this will garantee that only one of these 2 are being used at any given time. This could be a source of instability.
It's the same reason we have to use `file::withLock(path, function)` when reading/write to the filesystem, as the SD card also shares the SPI bus with the display on many devices.

Quite a lot of big changes have been pushed to the main repo in the past few weeks, so let me know if I can help with resolving merge conflicts.

d49406, 2025-09-21:
Hi Ken,

>The architecture looks solid to me.
Thank you! Great to hear that I managed to get the interface for something as vague as radios right.

>I'll have some minor remarks here and there, in an in-depth review.
e.g. RadiolibTactilityHal::spiXXTransaction() functions might need to use hal::spi::getLock().
Thanks for the hint. Currently I can pass either a lock object e.g. the LVGL lock, or a nullpointer which will use `hal::spi::getLock()` to get the bus lock for the host device.
I assumed I need the LVGL lock as it is also used in the SD card instance.

>Quite a lot of big changes have been pushed to the main repo in the past few weeks, so let me know if I can help with resolving merge conflicts.
Any help is appreciated, but I am certain this won't be a huge issue.

After reading CONTRIBUTING.md I realized that currently it isn't the time for new features.
So thank you for approving this new feature anyway.
I had a look at the bugs in ideas.md for next time, I think I'll have a look at the crash app next regarding encoder input.

PS: Github has an upload limit, thus I converted the vids down to 480p. Here are temporary links to the full vids:
RX - https://diskette.hoeglinger.eu/push/lora-rx.webm?k=O0e_GhK_2hu-YeTo
TX - https://diskette.hoeglinger.eu/push/lora-tx.webm?k=E0ZbIfdJO-HKMN9N

timoehl, 2025-09-21:
> This code uses the ESP32 RadioLib HAL layer by [@timoehl](https://github.com/timoehl) as a basis. Timoehl, is it okay to use parts of your implementation from this thread here and publish it under the GPLv3? [jgromes/RadioLib#1130](https://github.com/jgromes/RadioLib/discussions/1130)
> 
> Thank you by the way, this was a huge help on getting started.

@d49406 Completely fine with me, I am happy it could be of help.

KenVanHoeylandt, 2025-09-21:
> [@d49406](https://github.com/d49406) Completely fine with me, I am happy it could be of help.

Thank you @timoehl for the quick reply, and also for making RadioLib! I'm very excited that my project is getting radio functionality thanks to your and @d49406 's efforts.

KenVanHoeylandt, 2025-09-21:
Regarding radio licensing/permits: I agree. I don't think there's currently a risk, as I'm not selling devices.
It might be a hurdle in the future if one of our hardware sponsors want to bring out a device with Tactility as the default firmware, but I guess we can tackle that problem if/when that happens.

That said, I think it's easier to solve things like region selection, a "limit power" flag, and operator IDs on an OS level so that any radio app can opt in if they want to.

> As ChirpChatter doesn't really fit in with the UX of other apps, and since it more of a power-user tool I planned it to be an free external app.

A different UX is not a problem to me.  I'm still looking into good new patterns as we're kind of early into the Tactility project. I've only done so far what I thought felt right, and recently adapted it so it would work well on non-touch devices. The main question would be how/if it would work on smaller or larger screens.

That said, it's most welcome as an external app. The main limitation to external apps is the amount of exposed functions in `tt.init.cpp`, but I'll gladly expand those or accept PRs for that.

External apps are probably preferable at this point, as I'm getting tight on firmware storage space. Before your PR, we are at `2.2 MB` out of `3 MB` in the app partition. I can expand this to about `3.5 MB` until we're really getting into trouble with `4 MB` ROM devices. Perhaps those devices will need some features cut.

Additionally, I will likely refactor internal applications into external ones. I'm thinking of the `GPIO` and `Serial Monitor`. The calculator is already available as an external app. They probably don't impact firmware size much though.

Another thing I'm thinking of is perhaps making drivers or entire sub-systems as external binaries (e.g. GPS subsystem, external display via I2C, etc.)

> https://gist.github.com/d49406/a9c378fb84e4b9e6648b96c8e83bd0d6
> Maybe for the future, this could be an idea.

Nice pattern!
I've been thinking about root access / file permissions / etc., but I always conclude that the user will always be able to access everything as they can read out all the memory everywhere. Including the encryption keys for the WiFi password.

The only proper solution to this would be privilege separation, but that's in beta and only supports S3 and C3 SOCs:
https://github.com/espressif/esp-privilege-separation

KenVanHoeylandt, 2025-09-22:
Can you open `.gitmodules` and see if it contains this at the end:

```
[submodule "Libraries/minitar/minitar"]
	path = Libraries/minitar/minitar
	url = https://github.com/ByteWelder/minitar.git
```

If it's not there, it somehow disappeared in the merge and you can add the submodule manually with that URL and commit `78c254ba114f6b66d888149d4ad0eff178dceb88`.

If the submodule was already there:
Can you open a terminal at `Libraries/minitar/minitar` and run `git status`?
If the submodule is there as a git repo, you can run `git checkout 78c254ba114f6b66d888149d4ad0eff178dceb88` from that folder and it will give you the correct commit.

d49406, 2025-09-22:
It seems like the submodule isn't checked out. Weird.

```fish
n@baryonyx ~/w/p/Tactility>tail -n 20 .gitmodules                                                                                                                                         (2) 
[submodule "Libraries/FreeRTOS-Kernel"]
        path = Libraries/FreeRTOS-Kernel
        url = https://github.com/FreeRTOS/FreeRTOS-Kernel.git
[submodule "Libraries/mbedtls"]
        path = Libraries/mbedtls
        url = https://github.com/Mbed-TLS/mbedtls.git
[submodule "Libraries/lvgl"]
        path = Libraries/lvgl
        url = https://github.com/lvgl/lvgl.git
[submodule "Libraries/SDL"]
        path = Libraries/SDL
        url = https://github.com/libsdl-org/SDL.git
[submodule "Libraries/minitar/minitar"]
        path = Libraries/minitar/minitar
        url = https://github.com/ByteWelder/minitar.git
n@baryonyx ~/w/p/Tactility>cd Libraries/minitar/minitar
n@baryonyx ~/w/p/T/L/m/minitar>git status                                                                                                                                                 (1) 
On branch radio
Changes to be committed:
  (use "git restore --staged <file>..." to unstage)
        modified:   ../../../Boards/Simulator/Source/LvglTask.cpp
        modified:   ../../../TactilityCore/Source/EventFlag.cpp
        modified:   ../../../lv_conf.h

Unmerged paths:
  (use "git restore --staged <file>..." to unstage)
  (use "git add <file>..." to mark resolution)
        both modified:   ../../../Tactility/CMakeLists.txt

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
        modified:   ../../FreeRTOS-Kernel (new commits, modified content)
        modified:   ../../SDL (new commits)
        modified:   ../../lvgl (new commits)
        modified:   ../../mbedtls (new commits)
        modified:   ../../../TactilityCore/Source/EventFlag.cpp

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        ../../../runSim
        ../../../sdkconfig.bak

n@baryonyx ~/w/p/T/L/m/minitar>ls .git
ls: cannot access '.git': No such file or directory
```

Anyhow thanks for the hint, I reinitialized all submodules and it works fine now.
```
n@baryonyx ~/w/p/Tactility>git submodule update --init --recursive

Submodule 'Libraries/minitar/minitar' (https://github.com/ByteWelder/minitar.git) registered for path 'Libraries/minitar/minitar'
Cloning into '/home/n/work/private/Tactility/Libraries/minitar/minitar'...
Submodule path 'Libraries/FreeRTOS-Kernel': checked out 'def7d2df2b0506d3d249334974f51e427c17a41c'
Submodule path 'Libraries/FreeRTOS-Kernel/portable/ThirdParty/Community-Supported-Ports': checked out '4273ca3211b99914f31518378fb590fbff064953'
Submodule path 'Libraries/FreeRTOS-Kernel/portable/ThirdParty/Partner-Supported-Ports': checked out '3f9c99a682c5c796bb7eb89fd9c4385688fce27a'
Submodule path 'Libraries/SDL': checked out '52146cf067ae950b1d431a84766fc7d0501f536a'
Submodule path 'Libraries/lvgl': checked out 'c033a98afddd65aaafeebea625382a94020fe4a7'
Submodule path 'Libraries/mbedtls': checked out 'daca7a3979c22da155ec9dce49ab1abf3b65d3a9'
Submodule path 'Libraries/minitar/minitar': checked out '78c254ba114f6b66d888149d4ad0eff178dceb88'
n@baryonyx ~/w/p/Tactility>idf.py clean
Executing action: clean
Running cmake in directory /home/n/work/private/Tactility/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/n/.espressif/python_env/idf5.5_py3.13_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/n/work/private/Tactility"...

[... 32 lines cut ...]
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/n/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
NOTICE: Processing 18 dependencies:
...
-- Project sdkconfig file /home/n/work/private/Tactility/sdkconfig
Loading defaults file /home/n/work/private/Tactility/sdkconfig.defaults...
warning: unknown kconfig symbol 'LV_USE_USER_DATA' assigned to 'y' in /home/n/work/private/Tactility/sdkconfig.defaults
warning: unknown kconfig symbol 'LV_DISP_DEF_REFR_PERIOD' assigned to '10' in /home/n/work/private/Tactility/sdkconfig.defaults
-- Compiler supported targets: xtensa-esp-elf
-- App "Tactility" version: 0.6.0-dev
...
-- Configuring done (21.4s)
-- Generating done (0.8s)
-- Build files have been written to: /home/n/work/private/Tactility/build
Running ninja in directory /home/n/work/private/Tactility/build
Executing "ninja clean"...
[2/2] Cleaning all built files...
Cleaning... 0 files.
Done
```

d49406, 2025-09-24:
An update about code 22: I forgot the main function and friends, guess I had tunnel vision after all. Makes sense that it throws this error, once added there weren't any issues.

Besides, the tips you gave me worked out great, and the new workflow saves much hassle.

One thing I noticed, you shouldn't build the SDK multiple times, I noticed that the IDF version was required to be "5.55.55.5" instead of "5.5", I guess it concats this particular file.
Easy fix tough of deleting the botched release.

d49406, 2025-09-24:
>Would you mind clarifying? Was void main() just missing or was there more to it?

I just didn't copy and adapt main.cpp, so neither a main function nor app registration was included.
So the mapping will fail because it can't find the entry point.

Tough I have some feedback to share regarding the SDK.
First, the STL isn't available, which wasn't obvious to me at first.
Obviously if you use it, mangled symbols will be missing. But the error reporting is spot on so you can deduce this at glance.

Then secondly, some LVGL symbols are missing, I suppose you can't have them all but for now I added the grid layout ones.
Same with <cstdio> functions, there I added them all in bulk but I will clean it to only include string formatting related ones, file IO isn't needed I think.
It would be nice to note that any new TactilityC functions must be included in tt_init.cpp.

With these traps out of the way, it's actually quite nice developing for this platform.

KenVanHoeylandt, 2025-09-24:
Some feedback regarding the [diff](https://git.hoeglinger.eu/dominic/Tactiliest/compare/main...radio):

First of all: great job! The code was pretty easy to read. I also liked how the implementation was consistent with other Tactility code, including the usage of mutex and constexpr.

I see you added the app to `ExternalApps/` which I removed yesterday night. Feel free to either merge it to the main Tactility project or otherwise open a PR at the new [TactilityApps repository](https://github.com/ByteWelder/TactilityApps).

Some feedback/questions (it's all minor stuff):

- `RadiolibTactilityHal::delayMicroseconds()`: Can we use `tt::kernel::delayMicros()` or `ets_delay_us()` here?
- `Drivers/RadioLibCompat/Source/RadiolibThreadedDevice.cpp`: The `TAG` is likely too long and might not show up (not sure)
- `RadiolibThreadedDevice` header: first `private:` is probably not needed
- `Sx1262::setParameter()`, `getLoraParameter()`, etc: perhaps we should move the `return` statement at the end to the `default:` case? It could improve readability by a small amount. (I don't feel strongly about this)
- `__cxa_pure_virtual()`: Feel free to put this into the main firmware and link it up via `tt_init.cpp` (or leave it as-is, that's fine too!)
- `manifest.properties`: When you eventually update from `main`, this file format has simplified a lot. You can compare it with [this](https://github.com/ByteWelder/TactilityApps/blob/main/Apps/HelloWorld/manifest.properties) 
- `paths->getSystemPathLvgl("icon_msgs.png");` this has also changed with the newer SDKs: With `.app` packaging, you can add an `assets/` folder where `tactility.py` resides. This folder is stored in the `.app` package. `AppPaths.h` will have `getAssetsPath(childPath)` for you to use. There will also be a `getUserDataPath()` which will be used for storing configs etc. The user data path will be either on `/data` or `/sdcard` depending on what's available.

Great job on the TactilityC implementations too. Happy to see you fully documented it <3

I wonder what you think about the `tt_hal_radio_` prefix: should we remove the `_hal` part for all those HAL-related functions? Or would that be too risky to clash with something in the future?

d49406, 2025-09-25:
>First of all: great job! The code was pretty easy to read. I also liked how the implementation was consistent with other Tactility code, including the usage of mutex and constexpr.

That is entirely due to a very readable basis, I've seen so much worse. Not naming names, but a certain kernel comes to mind.

>I see you added the app to ExternalApps/ which I removed yesterday night. Feel free to either merge it to the main Tactility project or otherwise open a PR at the new [TactilityApps repository](https://github.com/ByteWelder/TactilityApps).

Thanks for the heads-up. I'll continue for a while without rebasing to main until I have both completed the Radio Terminal and tweaked Tactility to a usable state from an application programming standpoint.

Also thanks for the feedback, here is my response
1. Good suggestion, I'll implement it.
2. I'll shorten this class' name to ThreadRadioLibDev, or just shorten the tag a bit to "ThdRadioLibDev".
3. That's a brainworm from work, usually to ensure you can both break at the default label and that the function has only one entry and exit point. I'll take your suggestion to make it more uniform with the rest.
4. That function was from the STL IIRC, so it isn't strictly needed. I'll omit it for now.
5. I'll cross that bridge when I come to it, but good thing you mentioned it.
6. Great info, I planned to store presets somewhere, and this will come in handy for sure!

>Great job on the TactilityC implementations too. Happy to see you fully documented it <3

I despise undocumented APIs. Again I could rant about kernel programming all day but I'll not get into it.
So I try to document what's facing another programmer. I would have preferred to document some of the new radio HAL, but was unsure if and how to do it.

>I wonder what you think about the tt_hal_radio_ prefix: should we remove the _hal part for all those HAL-related functions? Or would that be too risky to clash with something in the future?

I find it nice that it maps to the actual namespace in C++, plus you can autocomplete easily/grep for all hal functions.
That saved me a few minutes debugging symbol export.
I'm in favor of letting it be. Also I am lazy :)

Some new developments and potential issues:
1. I had trouble using the `Str` classes format append function with a float, it told me a symbol was missing.
Tried adding it, but it couldn't be compiled. The missing name is `__extendsfdf2` to be precise.
The least logical step is to ram a soft float implementation into the app, but that did it for now.
I wonder if there's a better way?
2. I noticed some glitches while I have my T-Pager hooked up to my laptop. Sometimes the boot freezes at the statusbar, almost anytime after flashing the screen stays black. A reset usually does it. Maybe related to soft resets from IDF.
Also flashing is quite slow, and most significantly I cannot turn the device off when it's plugged into my machine.
3. Just a little feedback: The app doesn't close when you `brrr` it via the dev service. I almost thought I corrupted my app's close button, but it turns out they were just overlaid on top of each other.
4. I managed to make any lv_obj scrollable, and highlight its scrollbar when focused. This improves UX greatly on encoder devices. I'll push the current code if you like so you have a reference if you want to utilize it in your apps.

That's all for now. Have a nice evening!

KenVanHoeylandt, 2025-09-27:
FYI: I refactored parts in `elf_loader` in [this merged PR](https://github.com/ByteWelder/Tactility/pull/347/files).
I hope to get it [backported](https://github.com/espressif/esp-iot-solution/issues/587) into the official library.

It has a lot of new symbols related to the "soft float" GCC functions.
I've moved those into a [separate file](https://github.com/ByteWelder/Tactility/pull/347/files#diff-db6325a0ac88c59dca3891815928c758866a2b08c82a61b2f1811b1ce84a52b0) but I left the others in place so it's easier for you to resolve merge conflicts. 
In the future I will make separate lists of symbols for stdlib, LVGL, etc.

d49406, 2025-09-27:
Regarding making "Radio Terminal" (I renamed it) an official app, this will take longer than the current objective of mine to add radio support.

As it's main purpose right now is to test, and discover any issues with the radio HAL, it's currently sitting at over a thousand lines, many of which are plain ugly. So I'll publish it to my git first, and I'll make an issue later in the official apps repo once I am happy with the code after refactoring.

As for the state of the app, right now configuration of any modulation with its many different parameters works.
So good progress there.

Lastly, since I wanted to subscribe to state changes I refactored the PubSub mechanism [into its own class](https://git.hoeglinger.eu/dominic/Tactiliest/commit/4625f56c6eb3dac67f80ba88e5c4a558db8b3c78#diff-d57054258e28fc3fc77cce5032718be27ee87277).
Are variadic templates a no-go? If not, perhaps this class should be moved so it can be utilized by other classes.

KenVanHoeylandt, 2025-09-29:
> Regarding making "Radio Terminal" (I renamed it) an official app, this will take longer than the current objective of mine to add radio support.
>
> As it's main purpose right now is to test, and [...]

Take your time!

> Lastly, since I wanted to subscribe to state changes I refactored the PubSub mechanism [into its own [...]

Variadic templates are welcome. I think this is a useful improvement, so feel free to make it part of a future PR.

Sidenote: I've updated the local SDK release script, which releases to `releases/TactilitySDK-(platform)` instead of `releases/TactilitySDK`. There's an open PR with `tactility.py` improvements&fixes related to this. I want to get this merged asap. (gonna review it after posting this)

KenVanHoeylandt, 2025-09-29:
> It won't be part of the PR and just make these functions app-internal.

Feel free to add it to the PR if you want, though, cause it's welcome in the main code.
I was thinking it could belong somewhere in `Tactility/Source/lvgl/` and then also with a TactilityC binding.

> Thank you for the info. I will make sure to update sooner than later.

I just pushed a change to `Buildscripts/release-sdk-current.sh` in Tactility, and merged the new `tactility.py` changes (v2.3.3)
If `tactility.py` doesn't update yet via CDN (propagation takes a bit) then you can find it [here](https://raw.githubusercontent.com/ByteWelder/TactilityTool/refs/heads/main/tactility.py).

NellowTCS, 2025-10-30:
hi! 
just curious but any updates on this?

d49406, 2025-12-25:
I lived. Merry Christmas and Happy Holidays :)
Since I got some free time on my hands finally, I might as well start with an update.

## State of current goals

While developing the test app, I got increasingly unhappy with the way parameters are handled.
It does work rudimentary, but it requires knowledge of the device itself.
What do I mean? Well, I apparently chose to implement a somewhat oddball IC with the NRF24 when it comes to that,
having at least two parameters which got gaps between successive values which have a common label, e.g. -18, -12, -6, and 0dBm for power. To the programmer or end user, they have to keep a database of acceptable values or dig into RadioLib for valid values. Let's call that "Problem A".

That's one problem, but then there's another.
The NRF24 has a multi-pipe capability, subscribing for packets for different addresses, up to 6 at a time.
Currently, my API only has explicitly only one such channel. Could be acceptable in this case, but there's room for improvement. I'll refer to it as "Problem B".

Finally, there's a whole other glaring problem: GPIO abstraction.
For the T-Pager, the add-on module with the NRF24 uses the expansion module connector.
This becomes a problem since a vital pin is behind an I2C I/O expander (TX Enable).
I would have got the code ready to go if this pin were yet another ESP32 GPIO, but this throws me for a giant loop.
This I'll name "Problem C".

## What now?

I did some light research, and Android apparently has device information querying in-built since 13, formerly "Hardware Interface Definition Language" (HIDL) which is superseded by "Android Interface Definition Language" (AIDL).
I suspect the best way to solve all problems would be to have some queryable binary format which reflects the configurable scope of a given device and its current values. I'll call such a thing "Device Specification". This would solve Problem A and B for one, and could be a nice help with Problem C.

For the latter, the interface for generalized IO could be as follows: The HAL programmer derives from the abstract GpioDevice class, which has implementation specific methods for configuring and using IO plus the method, while stuff like value checks are done in the abstract class using the programmer provided Device Specification.

Radios could be the same deal, parameter discovery and their allowed values are handled and  encoded in the specification while the interface only has the usual getter and setters as already implemented.

Here's a sketch on how such a specification could look like for GPIO and radio devices,
as a python sketch: [dev_specs.py](https://github.com/user-attachments/files/24335770/dev_specs.py)

All of this needs a few extra helper classes tough to handle the specification format, so this approach might not be ideal.
On the other hand, I see that TactilityC is still a thing, out of necessity I suppose because bridging C++ code is really hard from loadable binaries. This approach of a binary serialization format would simplify this interfacing.

## Considerations for Device Specification Format

I had a look at the available options for the binary format. Human-readable is out of the question I presume because of performance, but that issue could be dismissed because querying Device Specifications shouldn't be a recurring task.
I am familiar with Protobuf, and I would rather not having seen the ratsnest of code it generates.
Plus serialization with schemas seems overkill for a seldomly executed task.

So I looked further into schema-less formats, there I see BSON as a viable format.
Looks alright, but before getting deeper into this I want to discuss it with you all.

## In short

 - Encountered problems with just adding the NRF24, namely lookup of valid parameter values, having more than one pipe and the "TX Enable" pin being behind an GPIO expander, thus I'll remove it from the scope of this issue
 - Drafted up a solution addressing all three issues in form of runtime device specification lookup

So after a few months of work and a few weeks of pausing, I hit a snag.
Please let me know what you think.

Thanks for listening, but remember to enjoy the holidays!

d49406, 2025-12-25:
## Bonus: User configurable device tree

Ever since LilyGO released their NRF24 module, I had some ideas on how to do a linux-style device tree configuration.
As this is an optional accessory, I thought having such a thing would be appropriate.
Since runtime is no constraint, and memory only slightly, I figured a human readable format would be nice to have.

Marrying it with generic device drivers tough has lead me to mock up such a configuration in YAML.
I know, but still a step up from working with DTSes.

Anyway here's an example I came up with a few months ago: [devicetree.txt](https://github.com/user-attachments/files/24335911/devicetree.txt)

Note the usage of references to make it a full graph to model dependencies or links between devices.

KenVanHoeylandt, 2026-01-11:
I made a DTS file parser with Lark in Python. I'm going to experiment with that.

KenVanHoeylandt, 2026-01-30:
First of all, the project has moved to its own "TactilityProject" GitHub space ("Tactility" was taken). The domain name is also changed to keep it consistent (tactility.one -> tactilityproject.org)

All devices now have a basic devicetree based on the new kernel code that is being written in `TactilityKernel/`.
There are kernel devices for: root, GPIO, and I2C. (the GPIO driver is untested)
There is also a kernel driver specifically for the T-Lora Pager, but it does nothing yet. It will later get the core init code for that device.

I still need to write documentation, but it's similar to the Zephyr/Linux setup.

Each device in `Devices/` has a `devicetree.yaml` file, like this:
https://github.com/TactilityProject/Tactility/blob/main/Devices/lilygo-tlora-pager/devicetree.yaml
It specifies where to look for more bindings (where other devicetree.yaml files can be found that it needs) and the root one like the one above also specifies a DTS file to load, which in this case is:
https://github.com/TactilityProject/Tactility/blob/main/Devices/lilygo-tlora-pager/lilygo%2Ctlora-pager.dts

A [driver](https://github.com/TactilityProject/Tactility/blob/main/TactilityKernel/Include/tactility/driver.h) has:
- an optional API (a struct with function pointers)
- an optional type (to make it easy to find all devices of a certain type, without knowing their `compatible` strings)
- declare their compatibility identifier (e.g. "esp32-i2c" for the ESP32's I2C Controller)

A [device](https://github.com/TactilityProject/Tactility/blob/main/TactilityKernel/Include/tactility/device.h) has:
- a name
- a configuration struct
- an optional parent

Devices can be bound to a driver. 
Drivers can set a `void* driver_state` on the device.

d49406, 2026-02-08:
First of all, I need to apologize for not being communicative.
Simply put, I've bit off more than I can chew currently.
Currently I do not have the capacity to finish the idea proper.
However, I hope it can serve as a good study on how to proceed, if anyone's willing to do it.

I really like the hardware interface, that's huge on its own and props for doing it!
YAML being included is a surprise, but in combination to a strict device tree I can see they'll both complement each other.

News on my local source code, the NRF24 transmits successfully but I stopped due to the need for configurable non-core GPIOs. In theory it should work just fine.

Let me know if I should clean up, rebase and push my current codebase to github if anyone needs it.